mbridak/not1mm #1: ERROR: No matching distribution found for not1mm

bplipschitz, 2023-05-11:
Running Ubuntu 20.04.6 LTS.
Python 3.8.10

After entering
`pip install -U not1mm`
I get the error in the subject line.  I would like to try this out, even though it is Beta.

Thanks, Jim KK0U

mbridak, 2023-05-11:
I believe the minimum Python version needed would be 3.9
I'll spin up a copy of 20.04 in a VM and see if I can get it to work with 3.8.10

bplipschitz, 2023-05-11:
Okidoke -- thanks!

mbridak, 2023-05-11:
Okay, I loaded 20.04. I tried pulling the source down from github and running it with Python 3.8 and not going anywhere.
So I tried apt installing python3.9. Got it to pull down with pip. But as it turns out I was too lenient, and it actually needs 3.10. So I'd have to say sorry it's not happening.

bplipschitz, 2023-05-11:
All good -- I've got a 22.04 around here somewhere, and I'll try it on
that.  It's just not my contesting computer

closed 2023-05-12 by mbridak