mbridak/not1mm #8: Don't start if locale is not en_US.UTF-8

rolnas, 2023-06-17:
Traceback (most recent call last):
  File "/home/rolnas/.local/bin/not1mm", line 5, in <module>
    from not1mm.__main__ import run
  File "/home/rolnas/.local/lib/python3.10/site-packages/not1mm/__main__.py", line 2274, in <module>
    window = MainWindow()
  File "/home/rolnas/.local/lib/python3.10/site-packages/not1mm/__main__.py", line 356, in __init__
    self.check_for_new_cty()
  File "/home/rolnas/.local/lib/python3.10/site-packages/not1mm/__main__.py", line 666, in check_for_new_cty
    update_available = cty.check_update()
  File "/home/rolnas/.local/lib/python3.10/site-packages/notctyparser/bigcty.py", line 181, in check_update
    datetime.strptime(date_str, "%d-%B-%Y"), "%Y%m%d"
  File "/usr/lib/python3.10/_strptime.py", line 568, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
  File "/usr/lib/python3.10/_strptime.py", line 349, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '26-May-2023' does not match format '%d-%B-%Y'

rolnas, 2023-06-17:
Start with env LC_ALL=en_US.UTF-8 not1mm works OK.

mbridak, 2023-06-17:
I'll have to noodle this a bit. I can't recreate it for some reason.

mbridak, 2023-06-17:
What does your LC_TIME normally contain?

rolnas, 2023-06-17:
lt_LT.UTF-8 or en_DK.UTF-8

mbridak, 2023-06-17:
Ok, I trapped the error raised by the notctyparser.
I'm opening an issue over on that repo for the issue.

rolnas, 2023-06-18:
After lastest update (23.6.17):
cty parser returned an error: %s time data '26-May-2023' does not match format '%d-%B-%Y'
but at least don't crash, but it doesn't use cty data.

mbridak, 2023-06-18:
Yes, but the actual error is being generated by the notctyparser library.
I've opened the issue in that repo.

mbridak, 2023-06-18:
Okay, I've fixed the problem with notctyparser, and pushed a new version to pypi.
I've pinned the new version number of notctyparser in the dependencies list of not1mm.
So, if you update not1mm it should download the new notctyparser as well.

Hope this fixes your issue.

mbridak, 2023-06-20:
resolved with upstream library.

closed 2023-06-20 by mbridak

rolnas, 2023-06-20:
Traceback (most recent call last):
  File "/home/rolnas/.local/bin/not1mm", line 5, in <module>
    from not1mm.__main__ import run
  File "/home/rolnas/.local/lib/python3.10/site-packages/not1mm/__main__.py", line 2316, in <module>
    window = MainWindow()
  File "/home/rolnas/.local/lib/python3.10/site-packages/not1mm/__main__.py", line 380, in __init__
    self.check_for_new_cty()
  File "/home/rolnas/.local/lib/python3.10/site-packages/not1mm/__main__.py", line 689, in check_for_new_cty
    cty = notctyparser.BigCty(WORKING_PATH + "/data/cty.json")
  File "/home/rolnas/.local/lib/python3.10/site-packages/notctyparser/bigcty.py", line 58, in __init__
    locale.setlocale(locale.LC_ALL, "en_US")
  File "/usr/lib/python3.10/locale.py", line 620, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting

Locale en_US doesn't exists, we should use en_US.UTF-8, but better solution would be don't depend on locale settings and save/exchange data in locale independent way.

Locale should be used to show representation of date in local convenient way, not to store data.

mbridak, 2023-06-20:
I'm not storing the data that's in question.
It's pulling an XML feed http://www.country-files.com/category/big-cty/feed/

which contains to a link to the latest available cty file.

```html
<link>https://www.country-files.com/big-cty-26-may-2023/</link>
```
I don't have a way to dictate the format of the date given at the end of the link.

rolnas, 2023-06-20:
In this case I would write a date parsing code, independing of locale settings.
That's because even en_US.UTF-8 locale could be missing on some installations.