mbridak/not1mm #10: Import ADIF files for data transfer

rolnas, 2023-06-18:
There is generation of ADIF files for export, but there are no way to import back data from another source (at least ADIF format).

mbridak, 2023-06-18:
True.
I don't know if it will.

DE8MSH, 2023-10-30:
Hi. I know you don't know. ;) But will it? It'd be a very nice feature. With check if we had QSO before. 👋🏻

cbspidi, 2024-06-25:
bumping an old issue, but...
a solution could be to implement an integration to qrz.com with API key
you could implement two handy functions 
- lookup in your logbook 
- lookup in qrz database for potential matches, this is more useful than first one

cbspidi, 2024-07-19:
@DE8MSH , in case you would still ike to have those contacts inside the `not1mm` I did it by exporting ADIF from QRZ then I opened it with ADIF master and exported the data into CSV file
Next, I wrote a quick Python script to export some(not all) data from the CSV file and SQL insert it into the SQLite DB file.

It was not a pretty process, but it got the data into the program, so I can enjoy knowing if I had previous contact.
The process I come up is not as good as I would like it to be,  that is due to me not knowing how is data structured inside the database, ie. what is expected and where and what needs to change / where and why.

I would happily share the python script I put together to do the above.

NikoKamtsioras, 2024-09-23:
can i have it?

NikoKamtsioras, 2024-09-23:
@DE8MSH Can i have it?

NikoKamtsioras, 2024-09-23:
@cbspidi Can i have it?

FastThenLeft, 2025-02-15:
Author - please resist putting in features that are NOT contesting features! During a contest, the only time I care if I worked you before is in the contest I'm in right now! This isn't a general logging program (I would have actually never put that feature in...). It's a contest program. There are several great general purpose logging programs that run on Linux already. They can use one of those. THIS however is the only decent contesting program available for Linux. Please keep it that way!

mbridak, 2025-02-23:
I think this could be better served not by making a program to inject ADIF data back into the database, but to rather create an already supported call history file. Which can then be easily loaded by an already existing mechanism.

[See this](https://github.com/mbridak/not1mm?tab=readme-ov-file#call-history-files)

So say @rolnas, @DE8MSH or @cbspidi want to use the general logging plugin, but see past contact info they've had on other programs like the contacts name or a remark from a past log. @cbspidi could modify his already written program to convert the ADIF to a slightly different CSV type file. The general logging plugin looks for the fields `Call`, `Name` and `UserText`. So if his program was slightly modified to add a header to the start of the CSV type file then output the wanted fields, you'd get something like:

```text
!!Order!!,Call,Name,UserText,
# Remark to let the user know it's meant for the general plug in.
K6GTE,MIKE,Programmer
K5TUX,RUSS,A podcaster
N2CQR,,I think his name is Bill
N6QW,Pete,Pasta Pete
```

Bada Bing Bada Boom.

Everyone's happy.

cbspidi, 2025-03-11:
Thank you @mbridak , I take the output is a data ready for the SQL injection, nice!

mbridak, 2025-03-11:
Well, I guess you could use it for that. But it's more to be used as a [call history file](https://github.com/mbridak/not1mm?tab=readme-ov-file#call-history-files). Come to think of it, I should probably list each contest and list which fields each one looks for...

closed 2025-03-18 by mbridak

rolnas, 2025-03-18:
This issue is not solved, because original problem is not solved. Import is important if we work (for e.g.) in multiple category or if a part of contest were in one program and other in anoher (that happens for e.g. in field days). It is normal to have import if there is export of data. Not everyone have knowledge to write a script to convert ADIF to SQL for injection to sqllite DB.