mbridak/not1mm #11: Cannot change RST to something 59+xx or other free text

rolnas, 2023-06-18:
For casual QSO mode at least we should allow write any values.
Yes we can change in log window and edit QSO, but it requires much more time.

mbridak, 2023-06-18:
This is a bit bizarre. It's somehow related to CAT control... At least a check for CAT control.

If you change CAT control to None. You can enter whatever you like in the RST field while using General logging.

```text
<QSO_DATE:8:d>20230618
<TIME_ON:4>1505
<CALL:5>K6GTE
<NAME:4>Mike
<MODE:2>CW
<FREQ:6>14.032
<RST_SENT:3>599
<RST_RCVD:17>I'm a pretty pony
<COMMENT:31>strange fellow who likes ponies
<EOR>
```

But as soon as CAT control it turned on, it forces it to 599...

Clearly a bug...

I'll get on it.

mbridak, 2023-06-18:
Okay, pushed changes. this should be fixed.

mbridak, 2023-06-20:
resolved.

closed 2023-06-20 by mbridak