mbridak/not1mm #12: Switch to direct hamlib

darvark, 2023-07-11:
Hi,
can we switch to use directly hamlib instead of fetching data from third party software like flrig? It should simplify cat communication and can be easily extended to add rotor support.
Only cons of such solution is that we have to configure hamlib from not1mm level, but still this would be much closer to n1mm than current solution.

mbridak, 2023-07-11:
I'm not opposed to it. I wouldn't do it as a replacement. More like a 3rd option. I personally prefer rigctld. I have it (rigctld) start up as a service in the background and never have to think about it.

mbridak, 2023-07-11:
Okay, I made a branch for it.

mbridak, 2023-07-11:
All the hamlib specific code should live in cat_interface.py
The settings should live in the same dialog as the other CAT settings.
I'll figure out the dialog.

mbridak, 2023-07-11:
Added initial fields to dialog.

[image]

hamlib_device QComboBox
hamlib_device QComboBox
hamlib_device QComboBox
update_devices QPushButton
usehamlib_radioButton QRadioButton

Some more will be needed I assume.

darvark, 2023-07-12:
Generally that's all. In case of any custom settings maybe in the future interface can be extended, but for now it's enough.

One additional question. Why do you query PTT status? Is it needed for anything? Python bindings for hamlib has very poor documentation and it simply won't work in some cases  (some rigs doesn't provide information about ptt so hamlib can't get that info)

mbridak, 2023-07-12:
Setting PTT state is needed for voice keying.
Polling PTT was/is used when n1mm radio state packets are sent, and will be used if I get around to so2r stuff.

You can just have it return False if it's not available.

darvark, 2023-07-22:
Strange, I'm applying your changes in configuration.ui file, rebuilding with rebuild.sh script, and still I can't see those new controls in configuration window. Am I missing something?

mbridak, 2023-07-22:
Hmm that is strange. I haven't hook up the controls to any code yet. No
laptop on my vacation. But you should still see them.

mbridak, 2023-07-28:
Ah, I think I know why your not seeing it. I think you haven't checked out and applied your changes to the branch created for the issue.

The branch should be called something like 12-switch-to-direct-hamlb

darvark, 2023-07-30:
I've made changes on my own branch and then applied your changes to configuration.ui (I've found only those changes in your branch), so still it should be visible since changes were applied to UI file

mbridak, 2024-11-14:
Stale

closed 2024-11-14 by mbridak