mbridak/not1mm #15: CW keyer window

rolnas, 2023-07-19:
**Is your feature request related to a problem? Please describe.**
Sometimes I need to send something out of macros.

**Describe the solution you'd like**
A separate window with entry to send automatically, when entered.

**Describe alternatives you've considered**
xlog and tlf have keyer interface.

**Additional context**
Nothing to add here.

mbridak, 2023-07-19:
Hi. Im away from my computer for about a week. I'll look at this when i get
home. I think what your asking is a pop up window with text entry field to
get sent as you type it.

mbridak, 2023-09-03:
Completed with the 23-9-2 update.

CTRL-SHIFT-K exposes a field to enter CW text.

When done press CTRL-SHIFT-K or the Enter key to hide the field.

closed 2023-09-03 by mbridak