mbridak/not1mm #16: Change frequency in bandmap

rolnas, 2023-07-19:
**Is your feature request related to a problem? Please describe.**
When I control remote station, sometimes I have only hamlib/rigctld access.

**Describe the solution you'd like**
Allow mouse click to bandmap and calculate frequency to change.

**Describe alternatives you've considered**
Enter frequency inside enter field works.

**Additional context**
None

mbridak, 2023-07-28:
Hi, Sorry for the delay. I was on vacation.

I'm not quite clear as to the issue or feature request you are making.
When a spot is clicked on in the bandmap, it should pass a message back to not1mm to change frequency to that of the clicked spot. If not1mm is connected to the radio via rigctld it should tune the radio to the clicked spot.

rolnas, 2023-07-28:
It's about facility to change frequency without clicking on spots, but between them or in free from spots frequencies.

mbridak, 2023-07-28:
Ah, I see. There is, unfortunately,  a fundamental problem with that.
Each spot is a clickable object with a callback event reporting back who was clicked, and by this virtue returning the frequency as one of it's properties which we can then send a message about.

I think a better thing would be a virtual VFO knob. Like a rotary encoder connected to a USB port.

rolnas, 2023-07-28:
Yes, I agree, I have DJ2GO2 midi keyboard with two big rotary encoders.
Works OK with quisk (SDR transceiver control software) and others.
not1mm could have some plugin architecture to make it work also.

mbridak, 2023-08-07:
Don't know if this helps, but this is what I got so far.

Made a branch for this.
Had a raspberry pico and a pretty decent encoder with no detents.
Hacked up some firmware for the pico in circuit python.
Made a cheesy applet to talk to the pico and the radio, insert magic here.
 
Still needs a bunch of work, like pulling the pico's serial device from preferences etc. but it's mostly there.

mbridak, 2023-08-07:
Okay, Made the VFO app auto detect the USB pico vfo.
It's stable enough to merge into main. It works for me so I'm closing this with the merge.
The Firmware for the pico can be found in source tree.

closed 2023-08-07 by mbridak