mbridak/not1mm #33: Log window crashes when entering an incorrect number

qsantos, 2023-09-03:
**Describe the bug**
When an incorrect number is entered for the frequency in the log window, it crashes, and can never be opened any more, even by restarting not1mm.

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure there is a log entry
2. In the menus, click on “Window”, then “Log Window”
3. Double-click on the frequency for one of the log entries to edit it
4. Type “123.456.789”
5. Press return to confirm

**Expected behavior**
Either the input should not be accepted, or it should be saved as-is, with optionally an indication that it is malformed (e.g. red color).

**Screenshots**
```
Traceback (most recent call last):
  File "/home/qsantos/.local/pipx/venvs/not1mm/lib/python3.11/site-packages/not1mm/logwindow.py", line 807, in <module>
    window = MainWindow()
             ^^^^^^^^^^^^
  File "/home/qsantos/.local/pipx/venvs/not1mm/lib/python3.11/site-packages/not1mm/logwindow.py", line 145, in __init__
    self.get_log()
  File "/home/qsantos/.local/pipx/venvs/not1mm/lib/python3.11/site-packages/not1mm/logwindow.py", line 487, in get_log
    QtWidgets.QTableWidgetItem(str(round(float(freq), 2))),
                                         ^^^^^^^^^^^
ValueError: could not convert string to float: '123.456.789'
```

**Desktop:**
 - OS: Debian
 - Version 12

**Additional context**
\-

mbridak, 2023-09-03:
Thanks for the report.
It should be fixed soon.

mbridak, 2023-09-03:
fixed with 23.9.3.1

closed 2023-09-03 by mbridak

qsantos, 2023-09-04:
I confirm that this is fixed. Thanks!