mbridak/not1mm #36: I cannot open the bandmap if I am editing bandmap.py in an editor

linustolke, 2023-09-10:
**Describe the bug**
The Bandmap window does not open until I have closed emacs editing not1mm/bandmap.py.

This is confusing to the point I had to read the source code to understand why it didn't open.

**To Reproduce**
Steps to reproduce the behavior:
1. Open not1mm/bandmap.py in an editor `emacs not1mm/bandmap.py`.
2. Start not1mm.
3. Select Windows => Bandmap
4. No window is opened.
5. Close emacs.
6. Select Windows => Bandmap
7. The bandmap window appears.

**Expected behavior**
I expect the opening of a window, not to be affected by some other process on the computer.

**Desktop (please complete the following information):**
 - OS: Linux (Ubuntu MATE)
 - Version 22

**Workaround
There is really no reason for having the emacs open while running not1mm so this is not a real problem.

mbridak, 2023-09-11:
Hi, it could be that there is an instance of it hung. not1mm checks if there is a process already running before launching another one. You could try:

```bash
ps ax | grep bandmap.py
```

Too see if there is one crashed but not closed in the background. Something like:

```bash
56114 ?        Sl     0:00 /usr/bin/python /home/mbridak/Nextcloud/dev/not1mm/not1mm/bandmap.py
```

If found kill the process and try again.

linustolke, 2023-09-11:
No, the `ps ax | grep bandmap.py` shows
```
1117487 pts/6    Sl     0:01 emacs not1mm/bandmap.py
1117520 pts/6    S+     0:00 grep --color=auto bandmap.py
```
and when exiting emacs, everything works again.

mbridak, 2023-11-13:
Pushed a change 23-11-13 that fixes this.

closed 2023-11-13 by mbridak