mbridak/not1mm #42: ADIF export stops after the 3rd QSO (CQWW SSB)

HB9HLI, 2023-10-30:
**Describe the bug**
I took part in the CQWW SSB past week-end and logged in with not1mm. I noticed that the ADIF export stops after the 3rd QSO. In order to generate the whole log, I had to open each recorded QSO and then save each entry. Once done for all QSOs, the ADIF export worked. 
I think it's great to be able to use software for contesting natively under Linux. Unfortunately I don't know Python well enough to help you.

**To Reproduce**
Steps to reproduce the behavior:
1. Enter personal datas and select CQWW SSB contest.
2. Enter some QSO, il thing about 20 QSO.
3. Export to ADIF.
4. See error on output file with a text editor.

**Desktop :**
Ubuntu 22.04 (Mate)
Python 3.10.12
Installation of not1mm via pip3.

73's, Dom HB9HLI

HB9HLI, 2023-10-30:
Hello, thanks for the quick reply. I've just tried on another machine with the same version of Ubuntu and I can't reproduce the problem.
I'll try later on the same machine and come back. Is there a debug mode and if so, how do I activate it?

HB9HLI, 2023-10-30:
Yes, it's a bit peculiar, maybe a "-d" option with an output to a file like not1mm.debug, that would be great for debugging.
I've now tested from the right machine with the DB I used for the CQWW and added a few entries to the log. Strangely enough, it works even though I struggled during the contest ;-) There must be a reason somewhere though. I'll keep looking and if I see this bug again, I'll come back. Leave this issue open for a few more days and let me know when you want to close it.
Thanks and best 73s

HB9HLI, 2023-11-02:
Hello, I did some testing last night and couldn't reproduce the bug I had during the CQWW. Even installing a fresh VM with Ubuntu 22.04, installed the minimum required, the bug never reproduced. It will remain a mystery. I think it's better to close this case.  73's Dom. HB9HLI

closed 2023-11-02 by HB9HLI

mbridak, 2023-11-02:
Okay, thanks checking into it.
73, Mike K6GTE