mbridak/not1mm #44: I build VFO knob /w Pico Pi, Circuit Py and 600ths rotator. But...

DE8MSH, 2023-11-02:
Hi.
Today I built my VFO knob /w Pico Pi, Circuit Py, your "code.py" and 600ths rotator. VFO window in !1MM works (no freezing, no errors). I use !1MM, FLRig, SCU-17. 

But: If I rotate on rotator nothing happens. I built it as described on your GH page. Hmmm... How can I check it? Is there a Py script for function test? Any cool ideas? Cat controll is working if I click on cluster callsign or rotate on FT-450D, or change QRG on FLRig.

mbridak, 2023-11-03:
Hi. If the VFO window pops up and shows the correct frequency from the radio, I'd have to guess bad wiring.

If it couldn't talk to the Pico, you'd get a popup telling you that it couldn't find it. So we know the Pico has the correct firmware and passing serial traffic.

If the correct current frequency shows in the VFO window it can talk to the radio.

That leaves the Pico not seeing the rotary encoder for some reason. So either the encoder is bad, or miswired.

mbridak, 2023-11-03:
Assuming the encoder is the same one as mine,

Red Wire to Pico pin 40 (VBUS)
Black Wire to Pico pin 38 (GND)
White Wire to Pico pin 29 (GP22)
Green Wire to Pico pin 27 (GP21)

DE8MSH, 2023-11-03:
Hello Mike. Now it works. I connected 0V line (black) instead of GND :| and had to copy your code.py to Pico. A raw copy of cody.py had space errors or something like that.

closed 2023-11-03 by DE8MSH