mbridak/not1mm #46: Exporting only  one call?

DE8MSH, 2023-11-12:
Hi.

I'm using 23.11.2. If I call "Generate ADIF" !1mm expotrs only one entry.

[image]

What am I doing wrong here?

mbridak, 2023-11-12:
Most likely nothing. Can you start the program from a terminal, then export the adif and see if it pops out an error in the terminal?

DE8MSH, 2023-11-12:
That:

`[19:38:51] DEBUG __main__ - generate_adif Line 2468: ******ADIF*****
Traceback (most recent call last):
  File "/home/superuser/.local/lib/python3.10/site-packages/not1mm/__main__.py", line 2469, in generate_adif
    self.contest.adif(self)
  File "/home/superuser/.local/lib/python3.10/site-packages/not1mm/plugins/general_logging.py", line 231, in adif
    if len(grid) > 1:
TypeError: object of type 'NoneType' has no len()
`

mbridak, 2023-11-12:
Ah okay, thanks. I'll fix it.

mbridak, 2023-11-12:
Okay, I've pushed an update. 
Hopefully this fixes the issue.

DE8MSH, 2023-11-12:
Okay. 'll give it a try tomorrow morning. PC's off. 11:49 pm here .

DE8MSH, 2023-11-13:
Seems to work now. Will test further.

closed 2023-11-13 by DE8MSH