mbridak/not1mm #47: Can't build !1mm from source....

DE8MSH, 2023-11-12:
Hi. This is the reply on this Linux:

`DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=21
DISTRIB_CODENAME=vanessa
DISTRIB_DESCRIPTION="Linux Mint 21 Vanessa"
NAME="Linux Mint"
VERSION="21 (Vanessa)"
ID=linuxmint
ID_LIKE="ubuntu debian"
PRETTY_NAME="Linux Mint 21"
VERSION_ID="21"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=vanessa
UBUNTU_CODENAME=jammy
cat: /etc/upstream-release: Ist ein Verzeichnis
`
>>>>>>>>>>>>>>>>>>>

`pip install --upgrade pip
pip install setuptools
pip install build
source rebuild.sh
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pip in /home/username/.local/lib/python3.10/site-packages (23.3.1)
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: setuptools in /home/username/.local/lib/python3.10/site-packages (68.2.2)
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: build in /home/username/.local/lib/python3.10/site-packages (1.0.3)
Requirement already satisfied: packaging>=19.0 in /usr/lib/python3/dist-packages (from build) (21.3)
Requirement already satisfied: pyproject_hooks in /home/username/.local/lib/python3.10/site-packages (from build) (1.0.0)
Requirement already satisfied: tomli>=1.1.0 in /home/username/.local/lib/python3.10/site-packages (from build) (2.0.1)
WARNING: Skipping not1mm as it is not installed.
rm: das Entfernen von 'dist/*' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
* Creating venv isolated environment...
* Installing packages in isolated environment... (setuptools)
* Getting build dependencies for sdist...
configuration error: `project.version` must be pep440
DESCRIPTION:
    The version of the project as supported by :pep:`440`.

GIVEN VALUE:
    "23...31"

OFFENDING RULE: 'format'

DEFINITION:
    {
        "type": "string",
        "format": "pep440"
    }
Traceback (most recent call last):
  File "/home/username/.local/lib/python3.10/site-packages/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
    main()
  File "/home/username/.local/lib/python3.10/site-packages/pyproject_hooks/_in_process/_in_process.py", line 335, in main
    json_out['return_val'] = hook(**hook_input['kwargs'])
  File "/home/username/.local/lib/python3.10/site-packages/pyproject_hooks/_in_process/_in_process.py", line 287, in get_requires_for_build_sdist
    return hook(config_settings)
  File "/tmp/build-env-79r2vwwd/lib/python3.10/site-packages/setuptools/build_meta.py", line 358, in get_requires_for_build_sdist
    return self._get_build_requires(config_settings, requirements=[])
  File "/tmp/build-env-79r2vwwd/lib/python3.10/site-packages/setuptools/build_meta.py", line 325, in _get_build_requires
    self.run_setup()
  File "/tmp/build-env-79r2vwwd/lib/python3.10/site-packages/setuptools/build_meta.py", line 341, in run_setup
    exec(code, locals())
  File "<string>", line 1, in <module>
  File "/tmp/build-env-79r2vwwd/lib/python3.10/site-packages/setuptools/__init__.py", line 103, in setup
    return distutils.core.setup(**attrs)
  File "/tmp/build-env-79r2vwwd/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 159, in setup
    dist.parse_config_files()
  File "/tmp/build-env-79r2vwwd/lib/python3.10/site-packages/setuptools/dist.py", line 653, in parse_config_files
    pyprojecttoml.apply_configuration(self, filename, ignore_option_errors)
  File "/tmp/build-env-79r2vwwd/lib/python3.10/site-packages/setuptools/config/pyprojecttoml.py", line 66, in apply_configuration
    config = read_configuration(filepath, True, ignore_option_errors, dist)
  File "/tmp/build-env-79r2vwwd/lib/python3.10/site-packages/setuptools/config/pyprojecttoml.py", line 127, in read_configuration
    validate(subset, filepath)
  File "/tmp/build-env-79r2vwwd/lib/python3.10/site-packages/setuptools/config/pyprojecttoml.py", line 55, in validate
    raise ValueError(f"{error}\n{summary}") from None
ValueError: invalid pyproject.toml config: `project.version`.
configuration error: `project.version` must be pep440

ERROR Backend subprocess exited when trying to invoke get_requires_for_build_sdist
Defaulting to user installation because normal site-packages is not writeable
Obtaining file:///home/username/coding/not1mm
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build editable did not run successfully.
  │ exit code: 1
  ╰─> [44 lines of output]
      configuration error: `project.version` must be pep440
      DESCRIPTION:
          The version of the project as supported by :pep:`440`.
      
      GIVEN VALUE:
          "23...31"
      
      OFFENDING RULE: 'format'
      
      DEFINITION:
          {
              "type": "string",
              "format": "pep440"
          }
      Traceback (most recent call last):
        File "/home/username/.local/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/username/.local/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/username/.local/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 132, in get_requires_for_build_editable
          return hook(config_settings)
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/build_meta.py", line 468, in get_requires_for_build_editable
          return self.get_requires_for_build_wheel(config_settings)
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/build_meta.py", line 355, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/build_meta.py", line 325, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/build_meta.py", line 341, in run_setup
          exec(code, locals())
        File "<string>", line 1, in <module>
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/core.py", line 159, in setup
          dist.parse_config_files()
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/dist.py", line 653, in parse_config_files
          pyprojecttoml.apply_configuration(self, filename, ignore_option_errors)
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/config/pyprojecttoml.py", line 66, in apply_configuration
          config = read_configuration(filepath, True, ignore_option_errors, dist)
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/config/pyprojecttoml.py", line 127, in read_configuration
          validate(subset, filepath)
        File "/tmp/pip-build-env-s_eqh7lp/overlay/local/lib/python3.10/dist-packages/setuptools/config/pyprojecttoml.py", line 55, in validate
          raise ValueError(f"{error}\n{summary}") from None
      ValueError: invalid pyproject.toml config: `project.version`.
      configuration error: `project.version` must be pep440
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build editable did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
testbook:~/coding/not1mm$ 
`

What to do? I'm not a professional virtual env user...

mbridak, 2023-11-12:
Looks like the update_version.sh bash script inserted a bad date. (I believe i fixed that 2 weeks ago.)

You'd look into 2 files.

pyproject.toml

and 

not1mm/lib/version.py

you'll find lines with something like:

version="23...31"

change it to:

version="23.10.31"

then run the rebuild script again.

DE8MSH, 2023-11-12:
Yes. Now it builds.

After entering not1mm in shell:

Traceback (most recent call last):
  File "/home/user/.local/bin/not1mm", line 5, in <module>
    from not1mm.__main__ import run
  File "/home/user/coding/not1mm/not1mm/__main__.py", line 64, in <module>
    WORKING_PATH = os.path.dirname(loader.get_filename())
AttributeError: '_NamespaceLoader' object has no attribute 'get_filename'

DE8MSH, 2023-11-12:
Okay. What I did now: I did git clone into /home/user/

The I did 

pip install --upgrade pip
pip install setuptools
pip install build
source rebuild.sh

After this it shows:

https://github.com/DE8MSH/pasties/blob/main/!1mm_1

Hmmm..... Any ideas?

DE8MSH, 2023-11-12:
Okay. Solution: I kicked all old stuff from directories. Now it compiled and installed.

Wow. Install procedure is like a wife when you didn't cleaned up the room even if she said it. LOL.

closed 2023-11-12 by DE8MSH