mbridak/not1mm #48: Lookup not working?

DE8MSH, 2023-11-12:
Hi.

If I enter the one or the other call: Lookup doesn't seems to be working.

[image]

[image]

[image]

It always shows same distances. But: OP 1 and OP 2 ware living in different areas.

What can I do to get real results?

Tnx dear OP.

DE8MSH, 2023-11-12:
EDIT: Error msg seems to be:

> Exception in thread Thread-4 (check_callsign2):
> Traceback (most recent call last):
>   File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.10/threading.py", line 953, in run
>     self._target(*self._args, **self._kwargs)
>   File "/home/user/.local/lib/python3.10/site-packages/not1mm/__main__.py", line 2239, in check_callsign2
>     theirgrid = response.get("grid")
> AttributeError: 'tuple' object has no attribute 'get'

DE8MSH, 2023-11-12:
EDIT²: I enabled full log (directory debug) to have a look at it by myself first....

DE8MSH, 2023-11-12:
Works. But does not work.

[image]

It gets data from HamQTH. But does not put data into fields...

mbridak, 2023-11-12:
That helps a-lot thanks.

mbridak, 2023-11-13:
I believe this is fixed.

DE8MSH, 2023-11-13:
Hmm... I did source rebuild. Result:

https://github.com/DE8MSH/pasties/blob/main/!1mm_2

I cannot start not1mm again. I killed old directory, git cloned again. Same error.

commit up to b6bc812 worked for me yesterday.

DE8MSH, 2023-11-13:
Okay. I had to 
rm -r ~/.local/lib/python3.10/site-packages/not1mm/
and
rm -r ~/.local/lib/python3.10/site-packages/not1mm-23.11.12.dist-info/
...

DE8MSH, 2023-11-13:
Seems to work now. I have HamQTH only right now. So not all call will be lookuped. I'll subscribe QRZ.

closed 2023-11-13 by DE8MSH