mbridak/not1mm #50: Cannot get dark mode with pipx install

rolnas, 2023-11-12:
**Describe the bug**
Cannot get dark mode with pipx install

**To Reproduce**
Steps to reproduce the behavior:
1. pipx install not1mm
2. run not1mm

**Expected behavior**
Dark mode

**Desktop (please complete the following information):**
 - OS: Debian
 - Version 12

**Additional context**
QT_STYLE_OVERRIDE="Adwaita-Dark" doesn't solve, because it is not installed inside pipx venv for not1mm.
Installed also adwaita-qt package in OS, but it looks it is not used inside pipx venv.

mbridak, 2023-11-12:
Yup, that's true. I have the same issue. Dark Mode is a real PITA.

rolnas, 2023-11-15:
Got it to work by manually creating directory `~/.local/pipx/venvs/not1mm/lib/python3.11/site-packages/PyQt5/Qt5/plugins/styles` and symlink `ln -s /usr/lib/x86_64-linux-gnu/qt5/plugins/styles/adwaita.so ~/.local/pipx/venvs/not1mm/lib/python3.11/site-packages/PyQt5/Qt5/plugins/styles/`.
File `/usr/lib/x86_64-linux-gnu/qt5/plugins/styles/adwaita.so` is from `adwaita-qt` package.