mbridak/not1mm #51: After clean build and entering new log db data 1st time: Exit /w error

DE8MSH, 2023-11-12:
Uhm....

user@user:~$ not1mm
[21:24:14] WARNING __main__ - <module> Line 2538: debugging off
Traceback (most recent call last):
  File "/home/user/not1mm/not1mm/__main__.py", line 1445, in save_contest
    self.load_contest()
  File "/home/user/not1mm/not1mm/__main__.py", line 803, in load_contest
    self.contest = doimp(self.contest_settings.get("ContestName"))
  File "/home/user/not1mm/not1mm/__main__.py", line 2508, in doimp
    return importlib.import_module(f"not1mm.plugins.{modname}")
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 879, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1017, in get_code
  File "<frozen importlib._bootstrap_external>", line 947, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/user/not1mm/not1mm/plugins/general_logging.py", line 277
    ...
    ^
IndentationError: expected an indented block after 'except' statement on line 276
Abgebrochen (Speicherabzug geschrieben)

HELP ME lol

DE8MSH, 2023-11-12:
Huh???

[image]

`...` is a command???

DE8MSH, 2023-11-12:
Ahhhh... Ur working on it right now?! Right?

DE8MSH, 2023-11-12:
Okay. I patched `...` into `print("jjjjj")`

closed 2023-11-12 by DE8MSH

mbridak, 2023-11-13:
Indentation error ... should have been indented one more level.