mbridak/not1mm #54: Add name if lookuped into other_1 field. But......

DE8MSH, 2023-11-13:
I added this line (green):

[image]

But it creates a fine segmentation fault. Uhm... Why?

mbridak, 2023-11-13:
That's a good question. I don't see a reason for it to segfault.
But I would caution you against making field setting code in __main__.py

Only because the fields other_1 and other_2 are often modified and read by the contest plugins. They serve different purposes based on the contest chosen. You could nest it in code that makes sure it's running the general logging plugin. something like:

if self.contest:
    if self.contest.name ==  "General Logging":
        self.other_1.triggerCrash(True)

But back to the original question.... I really don't know.

DE8MSH, 2023-11-13:
hmmmm. five lines deeper i use nick, too. no probs there. only if i settext to other_1 -- and i guess tabbing to name field (other_1) and focussing it. to me it makes no sense to xtract name/nickname and dont set it to other_1 (general logging)

mbridak, 2023-11-13:
Yeah it's not the nick variable.

I inserted:

```python
print(f"{type(self.other_1)}")
self.other_1.setText("test")
```

which resulted in:

```
<class 'PyQt5.QtWidgets.QLineEdit'>
Segmentation fault (core dumped)
```

So at the time self.other_1 is still a valid QLineEdit object, and I'm just passing in a plain vanilla string object. 
As far as I can tell it's crashing in the PyQt library.