mbridak/not1mm #57: No menu works after upgrade to 23.11.16

rolnas, 2023-11-17:
**Describe the bug**
No menu works after upgrade to 23.11.16, when I try to enter any menu entries (for e.g. Help) - 

**To Reproduce**
pipx upgrade no1mm:
1. Run not1mm
2. Click on 'Help'
3. Nothing appears

**Expected behavior**
Menu for Help should appear.

**Desktop (please complete the following information):**
 - OS: Debian
 - Version 12

**Additional context**
Any other menu entries also don't work.

mbridak, 2023-11-17:
Hi, I installed Debian 12 into a VM, installed not1mm with pipx and I'm not able to reproduce the problem.

rolnas, 2023-11-17:
Forgot to add, it is under wayland and with QT_QPA_PLATFORM=wayland - no menus.
The same under wayland but with QT_QPA_PLATFORM=xcb - menus works OK.
But previous versions worked with QT_QPA_PLATFORM=wayland, after some last changes - stopped to work.

rolnas, 2023-11-17:
Did some testing with older versions - works. Reinstalled newest version from scratch (by removing ~/.local/pipx) - started to work. Tried to upgrade from previous version - (but from clean previous install) - works. So probably some leftovers from even older install could cause those problems.

closed 2023-11-17 by rolnas