mbridak/not1mm #82: Bandmap and spots should work without

arodland, 2024-03-03:
**Is your feature request related to a problem? Please describe.**
If I'm operating unassisted, I still want to add people to my bandmap. Maybe I heard someone, but I can't work them right now, so I want to save them, spin the dial, and Ctrl-G them to try later. Maybe I just want to mark the station I *can't* work (like the domestic in the DX contest) so that I can spin past them next time.

**Describe the solution you'd like**
Either Ctrl-S should still add a spot to the bandmap when there's no telnet connection, or there could be another key to add to the local bandmap and send to the multicast group, without sending to the cluster.

mbridak, 2024-03-03:
Yes. I think something like this was suggested when I was on Linux in the Ham Shack. It sounded good them and equally good now. I'll work on it. Maybe CTRL-M for Mark.

mbridak, 2024-03-03:
Added feature as CTRL-M.

closed 2024-03-03 by mbridak