mbridak/not1mm #84: Add option for general VHF logging possibility (e.g DARC VHF Contests)

winnieXY, 2024-03-05:
**Is your feature request related to a problem? Please describe.**
No - it's a feature request.

**Describe the solution you'd like**
I would like to be able to use not1mm to log DARC style VHF/UHF contests where you get points depending on the distance between the contacts (1 point per km). The kilometers are calculated using the maidenhead locators which are exchanged during these contests.

Exchange is "RST number and locator. E.g. "59 016 JN49hg"

You find the contests e.g. here (sadly in english):
https://www.darc.de/der-club/referate/conteste/ukw/tnukwcontest000/tnukwcontest00700/

A possibility to calculate the distance between two maidenhead locators is e.g. shown here:
https://github.com/9V1KG/maidenhead/tree/master

mbridak, 2024-03-07:
Thanks. I'll take a look. I already have code in the repo to calc the distances between two given grid squares.

winnieXY, 2024-03-07:
I also started to work on a vhf / uhf logging plugin for you, but I'm not yet finished (and it's not yet tested) - so no MR yet :-)