mbridak/not1mm #86: CWR should be logged as CW

tcarwash, 2024-03-14:
**Summary**
QRZ and LOTW don't function correctly with log entries using CWR as mode instead of CW. 

**Description** 
Although functionally equivalent, QRZ doesn't consider CW == CWR and LOTW doesn't accept CWR as a valid mode at all. Rigctld reports CWR when some (e.g. IC7300) rigs are in CW-Reverse (upper sideband) mode, not1mm is logging this instead of just CW.

**Expected Behavior**
Some loggers handle this with a config setting like "report CWR as CW" It's probably just as fine to just change CWR into CW at some point before it's written to a file.

mbridak, 2024-03-15:
I'll push a fix soon thanks!

mbridak, 2024-03-15:
Okay, I pushed 24.3.15, hopefully fixing this.

tcarwash, 2024-03-15:
Awesome, thanks and 73!

closed 2024-03-16 by mbridak