mbridak/not1mm #104: Callsign lookup in QRZ database

cbspidi, 2024-06-26:
`not1mm` would be a very good general logger if there was a possibility to look up a callsign in the QRZ database.
This functionality helps a lot when working portable, and when reception is not so great, it helps validate the callsign and speeds the QSO by quite a bit.

**Describe the solution you'd like**
When typing the callsign we can have a special function to capture if the input ends with `?`, if yes run a wildcard search in QRZ database. 
The QRZ database will return all possible matches of the callsigns registered with qrz.com

Example:
User input in callsign box (you can create a validation there to check if a special character is used; it must be a minimum 4 characters long; otherwise, you will receive hundreds of callsigns)
`m0so?` or `m0so*`
we run qrz.com search for `m0so*`

Output:
Return from qrz is 15 possible matches, like the list below. 
```
M0SOT	Andrew Cowan Southampton
M6LZP	J Barry Lovelock The Lizard (alias match)
M0SOA	Gary McCourty Eaton, TARPORLEY
M0SOE	Bruce MacMillan Deal
M0SOO	Jason A Barker Scarborough
M0SOP	Harry Weston I Also Have The Call-sign M1ETU
M0SOX	GEL Galliver Odiham, HOOK
M0SOL	SOLWAY DX GROUP. C/O C.E.WOLF, G6LSO. CARLISLE
M0SOM	Mike Childs Chippenham
M0SOF	Standish Amateur Radio Society Standish
M0SOY	Gary Sawyer Tyne and Wear
M0SON	jason r i searle gosport
2E0SOY	Gary Sawyer Washington. Tyne and Wear (alias match)
M0SOC	Dr. Ryan Pike Nr. Aylesbury, Buckinghamshire
M7SOY	Gary Sawyer Washington. Tyne and Wear (alias match)
```

I have not found a logger for Linux that would have such great functionality 

**Additional context**
Add any other context or screenshots about the feature request here.

mbridak, 2024-06-26:
Hmm. I'll look into it. I just got back from vacation. I'll clear my
backlog of field day logger fixes, and then check if the qrz api supports
multiple or batch lookups.

cbspidi, 2024-06-28:
Hello, 
Thanks, 
I do not think the API will not support lookups; from what I can see, it only does logbook operations, not DB lookups.

Your way into it will be via HTTP request on the search 

Question?
How did you do the QRZ lookup on callsigns from contest tables, when field where you show grid locators etc ...
Do you log in to qrz.com already, I suppose you do to fetch that data

mbridak, 2024-06-28:
They have a lookup API. We use a generic library called requests to pass in
our/your username and password to their API URL. They spit back a session
key. We then use the session key to make lookup requests which returns the
data in an XML format.