mbridak/not1mm #105: App crashing sporadically when changing frequency

raufaser, 2024-07-13:
**Describe the bug**
The app **sometimes** crashes when switching bands, qrg or clicking on a station the band map.

**To Reproduce**
Since it is sporadically i would say. Use the app and see when it crashes ;)

**Expected behavior**
No crash

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Debian Linux
 - Version v24.7.12

**Additional context**
I am using it with flrig and an IC-7300.

```
Traceback (most recent call last):
  File "/home/jan/.local/lib/python3.11/site-packages/not1mm/__main__.py", line 2536, in watch_udp
    self.rig_control.set_vfo(int(vfo))
  File "/home/jan/.local/lib/python3.11/site-packages/not1mm/radio.py", line 92, in set_vfo
    self.cat.set_vfo(vfo)
  File "/home/jan/.local/lib/python3.11/site-packages/not1mm/lib/cat_interface.py", line 322, in set_vfo
    return self.__setvfo_flrig(freq)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jan/.local/lib/python3.11/site-packages/not1mm/lib/cat_interface.py", line 331, in __setvfo_flrig
    return self.server.rig.set_frequency(float(freq))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1122, in __call__
    return self.__send(self.__name, args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1464, in __request
    response = self.__transport.request(
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1166, in request
    return self.single_request(host, handler, request_body, verbose)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1178, in single_request
    http_conn = self.send_request(host, handler, request_body, verbose)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1284, in send_request
    connection.putrequest("POST", handler, skip_accept_encoding=True)
  File "/usr/lib/python3.11/http/client.py", line 1119, in putrequest
    raise CannotSendRequest(self.__state)
http.client.CannotSendRequest: Request-sent
Abgebrochen (Speicherabzug geschrieben)
```

Let me know, if you need more info.

raufaser, 2024-07-13:
It just happened again. I switched from 15m CW to 15m SSB. The radio switched frequency but it didn't change mode and filter.

```
Traceback (most recent call last):
  File "/home/jan/.local/lib/python3.11/site-packages/not1mm/radio.py", line 55, in run
    vfoa = self.cat.get_vfo()
           ^^^^^^^^^^^^^^^^^^
  File "/home/jan/.local/lib/python3.11/site-packages/not1mm/lib/cat_interface.py", line 125, in get_vfo
    vfo = self.__getvfo_flrig()
          ^^^^^^^^^^^^^^^^^^^^^
  File "/home/jan/.local/lib/python3.11/site-packages/not1mm/lib/cat_interface.py", line 136, in __getvfo_flrig
    return self.server.rig.get_vfo()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1122, in __call__
    return self.__send(self.__name, args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1464, in __request
    response = self.__transport.request(
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1166, in request
    return self.single_request(host, handler, request_body, verbose)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1178, in single_request
    http_conn = self.send_request(host, handler, request_body, verbose)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/xmlrpc/client.py", line 1284, in send_request
    connection.putrequest("POST", handler, skip_accept_encoding=True)
  File "/usr/lib/python3.11/http/client.py", line 1119, in putrequest
    raise CannotSendRequest(self.__state)
http.client.CannotSendRequest: Request-sent
Abgebrochen (Speicherabzug geschrieben)
```

mbridak, 2024-07-13:
Thanks. I'll switch over to using flrig and track it down.

mbridak, 2024-07-14:
Okay I pushed a change v24.7.13, I think it resolves the problem.
I'll leave the issue open for you to confirm.

raufaser, 2024-07-14:
I just updated and using the new version for the rest of the IARU HF Championship. Will give info,

raufaser, 2024-07-14:
I used the version for the rest of IARU HF Championship and some General Logging. No more crashes.
:)

closed 2024-07-14 by raufaser