mbridak/not1mm #107: notctyparser

hellmind, 2024-07-14:
Traceback (most recent call last):
  File "/usr/bin/not1mm", line 5, in <module>
    from not1mm.__main__ import run
  File "/usr/lib/python3.12/site-packages/not1mm/__main__.py", line 32, in <module>
    import notctyparser
ModuleNotFoundError: No module named 'notctyparser'

~ ❯ paru notctyparser
1 aur/python-notctyparser 23.6.21-1 [+0 ~0.00] [Installed]

whats  wrong?

mbridak, 2024-07-15:
I do not know. I've only had limited exposure to Arch based distros. The only one I've used was Manjaro, which I used:

```bash
pamac build not1mm-git
```

to install it.

The only reason that pops into my head as to why not1mm would not see notctyparser if it's installed would be if not1mm or notctyparser is installed into different virtual environments...

Curious if there is a notctyparser folder in /usr/lib/python3.12/site-packages/

hellmind, 2024-07-15:
The installation went fine (also with Pamac), but the problem continued. To make it work, I had to:

Clone the repository:
```
git clone https://github.com/mbridak/not1mm
cd not1mm
```

Create a build.sh file and execute it:
```
#!/bin/bash
source .venv/bin/activate
python3 -m build
pip install -e .
```

Create a run.sh file and run it:
```
#!/bin/bash
source .venv/bin/activate
not1mm
```

Without  venv:
```
python --version:
Python 3.12.4
```

```
ls /usr/lib/python3.12/site-
packages/ | grep not
drwxr-xr-x    - root 21 May 19:36 annotated_types
drwxr-xr-x    - root 21 May 19:36 annotated_types-0.7.0.dist-info
drwxr-xr-x    - root 15 May 14:30 pyinotify-0.9.6-py3.12.egg-info
.rw-r--r--  89k root 13 Apr 11:21 pyinotify.py
```

maybe there is some error on https://aur.archlinux.org/packages/python-notctyparser

mbridak, 2024-07-15:
Hard for me to say where the problem for that is. None of this is the normal way for it to be installed.
If your're running from a git clone, you should just have to:

```bash
source rebuild.sh
```

This is assuming you are in the project directory and have pip setuptools and build installed.

hellmind, 2024-07-15:
I must create a venv to build it because  arch/artix got a python external managed environment 
So I can't just install things using pip

```
not1mm on  master [?] is 󰏗 v24.7.13 via  v3.12.4 ➜ source rebuild.sh
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try 'pacman -S
    python-xyz', where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Arch-packaged Python package,
    create a virtual environment using 'python -m venv path/to/venv'.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-Arch packaged Python application,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. Make sure you have python-pipx
    installed via pacman.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
rebuild.sh (line 3): No matches for wildcard 'dist/*'. See `help wildcards-globbing`.
rm dist/*
   ^~~~~^
from sourcing file rebuild.sh
* Creating isolated environment: venv+pip...
* Installing packages in isolated environment:
  - setuptools
* Getting build dependencies for sdist...
running egg_info
writing not1mm.egg-info/PKG-INFO
writing dependency_links to not1mm.egg-info/dependency_links.txt
writing entry points to not1mm.egg-info/entry_points.txt
writing requirements to not1mm.egg-info/requires.txt
writing top-level names to not1mm.egg-info/top_level.txt
reading manifest file 'not1mm.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'not1mm.egg-info/SOURCES.txt'
* Building sdist...
running sdist
running egg_info
writing not1mm.egg-info/PKG-INFO
writing dependency_links to not1mm.egg-info/dependency_links.txt
writing entry points to not1mm.egg-info/entry_points.txt
writing requirements to not1mm.egg-info/requires.txt
writing top-level names to not1mm.egg-info/top_level.txt
reading manifest file 'not1mm.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'not1mm.egg-info/SOURCES.txt'
running check
creating not1mm-24.7.13
creating not1mm-24.7.13/not1mm
creating not1mm-24.7.13/not1mm.egg-info
creating not1mm-24.7.13/not1mm/data
creating not1mm-24.7.13/not1mm/data/phonetics
creating not1mm-24.7.13/not1mm/lib
creating not1mm-24.7.13/not1mm/plugins
copying files to not1mm-24.7.13...
copying LICENSE -> not1mm-24.7.13
copying README.md -> not1mm-24.7.13
copying pyproject.toml -> not1mm-24.7.13
copying not1mm/__init__.py -> not1mm-24.7.13/not1mm
copying not1mm/__main__.py -> not1mm-24.7.13/not1mm
copying not1mm/bandmap.py -> not1mm-24.7.13/not1mm
copying not1mm/checkwindow.py -> not1mm-24.7.13/not1mm
copying not1mm/fsutils.py -> not1mm-24.7.13/not1mm
copying not1mm/logwindow.py -> not1mm-24.7.13/not1mm
copying not1mm/radio.py -> not1mm-24.7.13/not1mm
[... 634 lines cut ...]
adding 'not1mm-24.7.13.dist-info/WHEEL'
adding 'not1mm-24.7.13.dist-info/entry_points.txt'
adding 'not1mm-24.7.13.dist-info/top_level.txt'
adding 'not1mm-24.7.13.dist-info/RECORD'
removing build/bdist.linux-x86_64/wheel
Successfully built not1mm-24.7.13.tar.gz and not1mm-24.7.13-py3-none-any.whl
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try 'pacman -S
    python-xyz', where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Arch-packaged Python package,
    create a virtual environment using 'python -m venv path/to/venv'.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-Arch packaged Python application,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. Make sure you have python-pipx
    installed via pacman.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

hellmind, 2024-07-15:
I must create a venv to build it because  arch/artix got a python external managed environment 
So I can't just install things using pip

```
not1mm on  master [?] is 󰏗 v24.7.13 via  v3.12.4 ➜ source rebuild.sh
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try 'pacman -S
    python-xyz', where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Arch-packaged Python package,
    create a virtual environment using 'python -m venv path/to/venv'.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-Arch packaged Python application,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. Make sure you have python-pipx
    installed via pacman.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
rebuild.sh (line 3): No matches for wildcard 'dist/*'. See `help wildcards-globbing`.
rm dist/*
   ^~~~~^
from sourcing file rebuild.sh
* Creating isolated environment: venv+pip...
* Installing packages in isolated environment:
  - setuptools
* Getting build dependencies for sdist...
running egg_info
writing not1mm.egg-info/PKG-INFO
writing dependency_links to not1mm.egg-info/dependency_links.txt
writing entry points to not1mm.egg-info/entry_points.txt
writing requirements to not1mm.egg-info/requires.txt
writing top-level names to not1mm.egg-info/top_level.txt
reading manifest file 'not1mm.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'not1mm.egg-info/SOURCES.txt'
* Building sdist...
running sdist
running egg_info
writing not1mm.egg-info/PKG-INFO
writing dependency_links to not1mm.egg-info/dependency_links.txt
writing entry points to not1mm.egg-info/entry_points.txt
writing requirements to not1mm.egg-info/requires.txt
writing top-level names to not1mm.egg-info/top_level.txt
reading manifest file 'not1mm.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'not1mm.egg-info/SOURCES.txt'
running check
creating not1mm-24.7.13
creating not1mm-24.7.13/not1mm
creating not1mm-24.7.13/not1mm.egg-info
creating not1mm-24.7.13/not1mm/data
creating not1mm-24.7.13/not1mm/data/phonetics
creating not1mm-24.7.13/not1mm/lib
creating not1mm-24.7.13/not1mm/plugins
copying files to not1mm-24.7.13...
copying LICENSE -> not1mm-24.7.13
copying README.md -> not1mm-24.7.13
copying pyproject.toml -> not1mm-24.7.13
copying not1mm/__init__.py -> not1mm-24.7.13/not1mm
copying not1mm/__main__.py -> not1mm-24.7.13/not1mm
copying not1mm/bandmap.py -> not1mm-24.7.13/not1mm
copying not1mm/checkwindow.py -> not1mm-24.7.13/not1mm
copying not1mm/fsutils.py -> not1mm-24.7.13/not1mm
copying not1mm/logwindow.py -> not1mm-24.7.13/not1mm
copying not1mm/radio.py -> not1mm-24.7.13/not1mm
[... 213 lines cut ...]
adding 'not1mm-24.7.13.dist-info/WHEEL'
adding 'not1mm-24.7.13.dist-info/entry_points.txt'
adding 'not1mm-24.7.13.dist-info/top_level.txt'
adding 'not1mm-24.7.13.dist-info/RECORD'
removing build/bdist.linux-x86_64/wheel
Successfully built not1mm-24.7.13.tar.gz and not1mm-24.7.13-py3-none-any.whl
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try 'pacman -S
    python-xyz', where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Arch-packaged Python package,
    create a virtual environment using 'python -m venv path/to/venv'.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-Arch packaged Python application,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. Make sure you have python-pipx
    installed via pacman.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

mbridak, 2024-07-15:
That's probably why i use pipx. It handles the venv automatically.

hellmind, 2024-07-16:
Nice, using pipx install not1mm works fine.

mbridak, 2024-07-16:
:thumbsup:

closed 2024-07-16 by mbridak