mbridak/not1mm #108: Can't select  select my radio soundcard

hellmind, 2024-07-16:
I'm using Artix (Arch without systemd) Linux with PipeWire. Despite selecting different output options, the audio always gets redirected to the default soundcard.

Expected behavior

The audio should be routed to the selected output device instead of the default soundcard.

mbridak, 2024-07-16:
Do you have libportaudio2 installed?

hellmind, 2024-07-16:
I got portaudio 1:19.7.0-2  installed ( A free, cross-platform, open source, audio I/O library.)
I don't see libportaudio2 on repository

mbridak, 2024-07-16:
Hmm. Might be okay. launch it from a terminal and see if you get a line that says "portaudio is not installed".

```python
try:
    import sounddevice as sd
except OSError as exception:
    print(exception)
    print("portaudio is not installed")
    sd = None
```

I'm guessing it's loading, 'cause your getting sound output. Just to the wrong location. I'll check it in about a week when my son is out of his soccer clinic. By the time I get home the sun has already baked the shack and it has no AC.

hellmind, 2024-07-22:
It seems to be wokring on latest version.

closed 2024-07-22 by hellmind