mbridak/not1mm #110: Accepting logs from GridTracker

cbspidi, 2024-07-16:
This software would be even better if it could accept log information from GridTracker, something like n1mm does. 
Grid Tracker can send logs as a forwarder or using a logging integration.

[image]

What do you think? Is this doable?

cbspidi, 2024-07-16:
Mega, 
Do you have plans on enabling it ?

mbridak, 2024-07-16:
I was going to do it for this last ARRL Field Day, but the wife forgot to check the contest calendar when she planned our trip to Idaho. So my motivation dwindled. It'll get done. Just not sure when.

cbspidi, 2024-07-17:
My brother and I have forked your code for QRZ database lookup code testing; he is a much much better Pythonist than I am, but if you were to point us to what and where needs to be done/changed, we could start to make some progress for your review later.

Ps.
On QRZ DB lookup, we have scraped DB into a file, and we are doing local lookups, It works quite nicely; more testing is to be done later this and next month.

cbspidi, 2024-09-02:
Hey @mbridak , 
Just a quick touch base to see if you had any time to think about this request.

Piotr

mbridak, 2024-09-02:
Yup, Been noodling on it.

I think I'll pass the FT8 result packet to the contest plugin if it has the logic to process it. That way I wouldn't need to make changes to any contest plugin where it wouldn't make sense. 

```python
if hasattr(self.current_contest, "ft8_catcher"):
    self.current_contest.ft8_catcher(ft8_result)
```

That way contests like WPX CW and SSB can ignore it.

What would be helpful would be some example packets from various contests. The only one I have is from field day.
So far it just watches for wsjtx qso packets and re-formats the UDP packet into a dictionary so I can process it.

```json
{
    'CALL': 'KE0OG',
    'GRIDSQUARE': 'DM10AT',
    'MODE': 'FT8',
    'RST_SENT': '',
    'RST_RCVD': '',
    'QSO_DATE': '20210329',
    'TIME_ON': '183213',
    'QSO_DATE_OFF': '20210329',
    'TIME_OFF': '183213',
    'BAND': '20M',
    'FREQ': '14.074754',
    'STATION_CALLSIGN': 'K6GTE',
    'MY_GRIDSQUARE': 'DM13AT',
    'CONTEST_ID': 'ARRL-FIELD-DAY',
    'SRX_STRING': '1D UT',
    'CLASS': '1D',
    'ARRL_SECT': 'UT'
}
```

cbspidi, 2024-09-02:
Hey @mbridak, 

Below is what I have picked up on Wireshark when logging QSO on the GridTracker when selected to output logs into external `n1mm`

The comms are done via local loopback 127.0.0.1 on UDP port 2237, 
In attachment, you can find the original pcap file 
[1loggedQSO-from-grid-tracker-to-n1mmlogger-udp-port2237.pcapng.zip](https://github.com/user-attachments/files/16840018/1loggedQSO-from-grid-tracker-to-n1mmlogger-udp-port2237.pcapng.zip).

What I can see in the packet, 

```
<CALL:5>OZ1BM,
<GRIDSQUARE:6>JO55GI,
<MODE:3>FT8,
<RST_SENT:3>+02,
<RST_RCVD:3>-15,
<QSO_DATE:8>20240902,
<TIME_ON:6>165700,
<QSO_DATE_OFF:8>20240902,
<TIME_OFF:6>165800,
<BAND:3>30m,
<FREQ:9>10.138706,
<STATION_CALLSIGN:5>G0TWP,
<MY_GRIDSQUARE:6>IO91SV,
<TX_PWR:2>25,
<COMMENT:25>FT8  Sent: +02  Rcvd: -15,
<DXCC:3>221,
<COUNTRY:7>Denmark,
<NAME:15>Brian Mikkelsen,
<EMAIL:13>brian@fkby.dk ,
<EOR>
```

mbridak, 2024-09-02:
I'm assuming it sends different different payloads for different contests.

For example the packet I mentioned above included some different tags. A couple specific to Field Day like the `CLASS` tag, and some more general contest ones like `SRX_STRING`.

```text
    'CONTEST_ID': 'ARRL-FIELD-DAY',
    'SRX_STRING': '1D UT',
    'CLASS': '1D',
    'ARRL_SECT': 'UT'
```

And are there contests that accept FT8 contacts that need a serial number? If so how does wsjtx get the S/N to send?

cbspidi, 2024-09-02:
Hmm, 
How do you have your wsjtx configured, do you remember or maybe have screengrabs so I can replicate and capture packets for you ?
Below is how I have it for special operating activity:

[image]

[image]

mbridak, 2024-09-02:
@cbspidi It might end up okay. I looked each of the contests I could find from your screenshot, and the only one mentioning a S/N was the FT Round Up. Since it's FT only, I think I can assume wsjtx will keep track of it's own S/N.

mbridak, 2024-09-03:
@cbspidi Okay, as a proof of concept I've added FT8/FT4 contacts to the ARRL Field Day contest plugin.

I used a small program I wrote in the testing folder to generate the WSJT/X Field Day QSO UDP packets.
[wsjtx_inject_udp.py](https://github.com/mbridak/not1mm/blob/master/not1mm/testing/wsjtx_inject_udp.py)

This is assuming WSJT/X will be sending packets to localhost:2237.

Cheers :tada:

cbspidi, 2024-09-04:
Hey
Good news,
Should I test it by pulling your repo and starting new contest log as "ARRL
Field Day" ?

cbspidi, 2024-09-05:
Hey @mbridak , 
I need a bit of help on testing this functionality. I have tried to push logs to 127.0.0.1 on port 2237, but nothing gets populated in my not1mm log; where do I go wrong ?

mbridak, 2024-09-05:
Update your install, I pushed some code bits to add some debug info.

then:

```bash
touch debug
not1mm
```

With WSJT running you should start seeing status packets like:

```text
[2024-09-05 11:25:13,559] DEBUG not1mm.lib.ft8_watcher - on_udp_socket_ready_read Line 71: datagram=b'\xad\xbc\xcb\xda\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x06WSJT-X\x00\x00\x00\x00\x00k\xf0\xd0\x00\x00\x00\x03FT8\x00\x00\x00\x05K5TUX\x00\x00\x00\x03-15\x00\x00\x00\x03FT8\x00\x00\x00\x00\x00\x02\x81\x00\x00\x02\x81\x00\x00\x00\x05K6GTE\x00\x00\x00\x06DM13AT\x00\x00\x00\x04DM13\x00\xff\xff\xff\xff\x00\x03\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x07Default\xff\xff\xff\xff'
Status: sv:2 p:1 u:WSJT-X df:7074000 m:FT8 dxc:K5TUX
```

It will only show QSO's if you have ARRL Field Day selected as your current contest.

Not sure how you'll generate Field Day contacts from WSJT without others to connect to who are also running WSJT in field day contest mode. That's why I used that testing program to generate the packets.

Oh I also made a FlDigi interface that will also work with the Field Day plugin. That one just makes repeated XMLRPC requests to fldigi asking for the last logged qso and updates not1mms log if its different than the last one it saw.

I have to get around to making confguration dialogs to be able to select non standard ports etc.

Then start adding them to more contests.

cbspidi, 2024-09-05:
Right, Roger, as you said I will need other station with ARRL field day on
qso with me for testing.
How would I go about enabling this functionality for general logging?

mbridak, 2024-09-05:
In each contest plugin where it's needed, we'd have to add an ft8_handler() function.

The Field Day one looks like:

```python
def ft8_handler(the_packet: dict):
    """Process FT8 QSO packets
    FT8
    {
        'CALL': 'KE0OG',
        'GRIDSQUARE': 'DM10AT',
        'MODE': 'FT8',
        'RST_SENT': '',
        'RST_RCVD': '',
        'QSO_DATE': '20210329',
        'TIME_ON': '183213',
        'QSO_DATE_OFF': '20210329',
        'TIME_OFF': '183213',
        'BAND': '20M',
        'FREQ': '14.074754',
        'STATION_CALLSIGN': 'K6GTE',
        'MY_GRIDSQUARE': 'DM13AT',
        'CONTEST_ID': 'ARRL-FIELD-DAY',
        'SRX_STRING': '1D UT',
        'CLASS': '1D',
        'ARRL_SECT': 'UT'
    }
    FlDigi
    {
            'FREQ': '7.029500',
            'CALL': 'DL2DSL',
            'MODE': 'RTTY',
            'NAME': 'BOB',
            'QSO_DATE': '20240904',
            'QSO_DATE_OFF': '20240904',
            'TIME_OFF': '212825',
            'TIME_ON': '212800',
            'RST_RCVD': '599',
            'RST_SENT': '599',
            'BAND': '40M',
            'COUNTRY': 'FED. REP. OF GERMANY',
            'CQZ': '14',
            'STX': '000',
            'STX_STRING': '1D ORG',
            'CLASS': '1D',
            'ARRL_SECT': 'DX',
            'TX_PWR': '0',
            'OPERATOR': 'K6GTE',
            'STATION_CALLSIGN': 'K6GTE',
            'MY_GRIDSQUARE': 'DM13AT',
            'MY_CITY': 'ANAHEIM, CA',
            'MY_STATE': 'CA'
        }

    """
    logger.debug(f"{the_packet=}")
    if ALTEREGO is not None:
        ALTEREGO.callsign.setText(the_packet.get("CALL"))
        ALTEREGO.contact["Call"] = the_packet.get("CALL", "")
        ALTEREGO.contact["SNT"] = ALTEREGO.sent.text()
        ALTEREGO.contact["RCV"] = ALTEREGO.receive.text()
        ALTEREGO.contact["Exchange1"] = the_packet.get("CLASS", "ERR")
        ALTEREGO.contact["Sect"] = the_packet.get("ARRL_SECT", "ERR")
        ALTEREGO.contact["Mode"] = the_packet.get("MODE", "ERR")
        ALTEREGO.contact["Freq"] = round(float(the_packet.get("FREQ", "0.0")) * 1000, 2)
        ALTEREGO.contact["QSXFreq"] = round(
            float(the_packet.get("FREQ", "0.0")) * 1000, 2
        )
        ALTEREGO.contact["Band"] = get_logged_band(
            str(int(float(the_packet.get("FREQ", "0.0")) * 1000000))
        )
        logger.debug(f"{ALTEREGO.contact=}")
        ALTEREGO.other_1.setText(the_packet.get("CLASS", "ERR"))
        ALTEREGO.other_2.setText(the_packet.get("ARRL_SECT", "ERR"))
        ALTEREGO.save_contact()
```

mbridak, 2024-09-10:
Closing as complete.

closed 2024-09-10 by mbridak

cbspidi, 2024-09-11:
Hey, 
I have found that not all programs send a header that you expect to see (magic number); grid-tracker does not, wsjtx also when you configure it as a secondary server will not send a header in the packet.

Also, have you, by chance, done ft8 for all content types?

Piotr

mbridak, 2024-09-11:
@cbspidi I've changed the FT8_watcher class to watch for traffic on the WSJT-X Multicast address of 224.0.0.1 port 2237.

Not sure what you mean by ft8 for all content types.

I still have things I need to do with it. I'd like it to get decodes and check the calls against already worked, then pass that back to wsjt-x to color code the calls for dupes. Also pass those calls to the bandmap.

cbspidi, 2024-09-11:
Ha, sorry autocomplete error, content= contest, i was under impression you did it only for ARRL field day

mbridak, 2024-09-11:
Ah okay, So far just the ARRL Field Day, the 3 ARRL VHF contest and the Weekly RTTY.

cbspidi, 2024-09-11:
Hello
Would it be much work to put it in the general logging also? I could do much more testing there.