mbridak/not1mm #111: Window position / location is not saved on exit

cbspidi, 2024-07-16:
**Describe the bug**
Each time the program is started, the position of the windows is reset to default.

For example, if I move `VFO Window` to whatever my preferred position is on exit, I will lose all the positioning I have changed, and on the next program start, everything will be reset to the default positioning.

**To Reproduce**
Move any of the detachable windows 
Exit the program
Start the program 

**Expected behaviour**
The positioning of the windows should be saved across program restarts

**Desktop (please complete the following information):**
 - OS: Linux Arch
 - Version 24.6.1

cbspidi, 2024-07-16:
O nooo ....
Hmm, I wonder, could we use the XML config file to save positioning upon exit ?

mbridak, 2024-07-16:
They are saved. Here's part of the of the not1mm.json file that is saved in home/.config/not1mm/

```json
    "window_height": 440,
    "window_width": 1252,
    "window_x": -11,
    "window_y": -49,
```

If they ever fix it, I'll include the same for the dockable windows. But until then it's wasted effort.

cbspidi, 2024-07-17:
Hmm, 
I do not use Wayland; I'm on X11(arch linux with xfce4).
We may not be talking about the same thing here.

I do not see any specific settings saved in jason file for each detachable window, 
Take a look at a quick video; maybe this will describe the issue better. 

https://github.com/user-attachments/assets/b75db726-c91e-45f5-9eac-7be624946172

cbspidi, 2024-07-17:
Could this be related to saving the geometry of the window ?

As described 
here 
[stackoverflow_link](https://stackoverflow.com/questions/14288635/any-easy-way-to-store-dock-windows-layout-and-sizes-in-settings-with-qt)

and 
here 
[gt_forum](https://forum.qt.io/topic/124207/save-position-and-size-of-the-application-window-and-load-it-for-the-next-time)

mbridak, 2024-07-17:
Ah, okay you'd like the docked positions saved left, right, top and bottom
relative to the main screen and not x/y screen location.
That is probably doable.

cbspidi, 2024-07-17:
Yes, docked positions, it would be very good to have this saved on exit

closed 2024-07-20 by mbridak