mbridak/not1mm #112: Update qrz.com logbook on exit

cbspidi, 2024-07-19:
**Is your feature request related to a problem? Please describe.**
It would be a great functionality to have either a manual update button for QRZ updates or an option to select "update on exit."

That way, you would not need to worry about exporting whole logs into a `adif` file and manually updating your online platform.

**Describe the solution you'd like**
QRZ provides an API endpoint for updating your logbook, it sounds relatively simple to write a logic to do following.

1.  in local file, mark a timestamp of the last useful update 
2. when the update button is pressed (or the option update on exit is selected), check the time difference between now and the last successful update.
3. select all the records from the time window selected in point 2
4. update QRZ with the records from point 3
5. mark the new time stamp of the successful update

**Describe alternatives you've considered**
manual export and import to the online platform, painful if you use the program on a daily basis 

**Additional context**

FastThenLeft, 2025-02-15:
I would vote NO on this! At the end of the contest, just upload your log to QRZ. Next, someone will want to upload to eQSL and then ClubLog, and then HamQTH... The list would go on forever! Too many features is as bad as not enough. The program outputs ADIF and Cabrillo - that's all that's needed. The user can then do what he wants with those files. At the end of a 48-hour contest it would take one minute more to manually upload your log to QRZ. We don't need that functionality in this app!

I'd rather have the author concentrate on *contest features*!

mbridak, 2025-02-23:
Agreed.

closed 2025-02-23 by mbridak