mbridak/not1mm #113: Add more bands for quick band shifting (4m, 12m, 17m, 30m)

cbspidi, 2024-07-19:
**Describe the solution you'd like**

Please add more bands under configuration settings/bands (which will have an effect on selecting the operating band (next to the callsign window).

Please add the below bands:
4m, 
12m
17m 
30m

**Describe alternatives you've considered**
Manual band swap on the radio

mbridak, 2024-07-19:
Those are WARC bands. No contesting is allowed.

cbspidi, 2024-07-19:
Yeah, I get your point. 
I use your program (the best I have come across to this point) as a general logger.
With the above, I have QSOs on the mentioned bands in SSB mode, just not contest mode.

mbridak, 2024-07-19:
> in contest mode.

Have you tried QLog?
It's a way better general logger.

cbspidi, 2024-07-19:
Okay, thanks 

I will have a look(no i have not tried it before), what I like about your take on  login is
its simple
its all in python 
cross-platform 
very portable