mbridak/not1mm #114: CW operation via flrig

raufaser, 2024-07-19:
**Is your feature request related to a problem? Please describe.**
It is not possible to operate CW through flrig connection.

**Describe the solution you'd like**
Make it possible to operate CW through flrig connection.

**Additional context**
Excerpt from flrig manual
```
rig.cwio_set_wpm          n:i  set cwio WPM
rig.cwio_text             i:s  send text via cwio interface
rig.cwio_send             n:i  cwio transmit 1/0 (on/off)
```

command lines to test the CW API via XMLRPC

_Setting WPM_
`curl -d "<?xml version='1.0'?><methodCall><methodName>rig.cwio_set_wpm</methodName><params><param><value><i4>28</i4></value></param></params></methodCall>" http://localhost:12345
`

_Setting the text to send_
`curl -d "<?xml version='1.0'?><methodCall><methodName>rig.cwio_text</methodName><params><param><value><string>test test test</string></value></param></params></methodCall>" http://localhost:12345`

_Enable send_
`curl -d "<?xml version='1.0'?><methodCall><methodName>rig.cwio_send</methodName><params><param><value><i4>1</i4></value></param></params></methodCall>" http://localhost:12345`

mbridak, 2024-07-19:
Thanks I'll take a look.

mbridak, 2024-07-25:
Okay. I merged a code snippet. It might work... Probably not though. I tried to get flrig to send CW to 3 different radios I have to test with. I couldn't get it to send CW using it's own keyer interface to any of them. The most I can say about the code is that while testing, it didn't crash. So it didn't make it any worse.

raufaser, 2024-07-28:
It kinda works. 

In case the Send button is already yellow the text is not send. Only when you push another button. It can also be the "empty" button.

Please see the video to get an idea what i mean.

[1.webm](https://github.com/user-attachments/assets/13e2277d-66be-4646-949f-68d9b4410c41)

mbridak, 2024-07-28:
Okay. So If I understand correctly, it would work if I:

#1 Turn on the send button when the mode is changed to CW.
#2 Just send the text, and it should be managed by the FLRIG's internal text buffer.
#3 if mode changed to something not CW, turn off the send button.

Also:

If the mode is CW, I can cycle the send button to clear the buffer if the user presses the ESC key in the not1mm interface.

mbridak, 2024-07-29:
Okay pushed a try # 2.
Go ahead and give it a try.

raufaser, 2024-07-29:
It works.
Only thing i noticed. If you manually disable the Send/Pause button in the flrig cw interface, then nothing is send when you hit a macro button. The text buffer is filled until you either manualky reenable the Send/Pause button or switch to SSB and back to CW. 
I guess best is not to manually fiddle around with the CW interface when you use Not1MM to send CW :)

What is not working: Setting WPM

Thanks so far!

mbridak, 2024-07-29:
Yeah, I looked through their xmlrpc docs, and they have no way to query the state of the send button. So I have no way to know if someone clicked it on their own.

mbridak, 2024-07-29:
Okay. Added setting the speed.

closed 2024-07-31 by mbridak