mbridak/not1mm #120: ESM and CQ Auto

py4xx, 2024-07-26:
**Is your feature request related to a problem? Please describe.**
It's not really a problem, but the lack of a function, which for many competitors, is essential.

**Describe the solution you'd like**
ESM mode involves using the <ENTER> key to send the message before saving it to the log. In Not1MM, every time I am in the callsign field and press enter, it saves the callsign without saving the exchange. When in runner, when pressing <ENTER> in the callsign field, it could send the CALL + RST without saving the data yet. And when in the exchange field, once completed, it could send the TU message and only then save the data to the log. When in S&P, with the cursor in the callsign field, pressing <ENTER> sends {MYCALL} without saving it to the log, and when filling in the exchange field, pressing <ENTER> sends my exchange and saves it to the log.
CQ Auto allows us to send CQ without having to keep pressing F1 until the callsign field is filled in, interrupting CQ Auto. Pressing F1 again would return to CQ Auto. This could be done as a small rectangle option (attached), where when clicked, the CQ auto  function is activated and when unclicked, the function is deactivated.

**Describe alternatives you've considered**

**Additional context**
[image]

 Carlos - PY4XX

awandahl, 2024-09-19:
I second this enhancement. I would say it is necessary. To be able to attract the big guns, or the small pistols with some ambitions, not1mm has to be somewhat keyboard centric (as DXLog call it). To minimize the use of the mouse and pressing of the different keys as an option will be paramount. The logic of ESM is described well by Carlos above. Other sources are DXLog: https://dxlog.net/docs/index.php/Menu_Tools and N1MM: https://n1mmwp.hamdocs.com/setup/function-keys/#esm-enter-sends-message To me (a small pistol with some ambitions) I can't use not1mm properly in a contest as is now. I'm frequently guest operator at some big gun's operations and we have moved from N1MM+ to DXLog because of these small gains you can make by limiting the use of mouse/key clicks. I would love not1mm to be a tool for serious contest operating. That said, I realize the challenges in programming these features. No rush, other features may have priority. But in the end the ESM feature is 100% necessary.

mbridak, 2024-09-20:
Just a quick note. 
I don't think you have to do any mouse clicking as it is now. 
Everything can be done via the keyboard.

But I'll still think about it.

mbridak, 2024-10-11:
Done.

closed 2024-10-11 by mbridak