mbridak/not1mm #121: Call History

py4xx, 2024-07-26:
**Is your feature request related to a problem? Please describe.**
Using call History helps a lot in the competition.

**Describe the solution you'd like**
A way to create or import a CALL HISTORY file. 
When choosing the contest, not1mm automatically imports the contest's call history.
When filling in the CALLSIGN field, automatically, in the EXCH field, it is completed with the exchange of the station entered in the CALLSIGN field. 

**Describe alternatives you've considered**

**Additional context**
[RAEM.txt](https://github.com/user-attachments/files/16393457/RAEM.txt)

mbridak, 2024-10-30:
Okay this is the next Fun thing.

py4xx, 2024-10-30:
TNX... 73

mbridak, 2024-11-03:
Hello @py4xx, I'm pretty far along in making this work.

So far I'm making it work with N1MM style call history files I've downloaded from their site that start with a header line defining how the info maps to the database table. For example you'd have to add:

```text
!!Order!!, Call, Exch1,
```

Added as the first line of the RAEM.txt file you referenced in the your initial message.

I'll need to make a RAEM contest plugin, That's kind of a strange contest. Certainly has an odd point/multiplier system.

mbridak, 2024-11-03:
Okay, I couldn't sleep. So I added a RAEM contest plugin. Got the Call History, ESM, crazy Points and Mult system working with it. 

[image]

You'll have to be the bug hunter for this.

To Start:

Modify your RAEM call history file and add this header:

[image]

Then go to `File -> Load Call History File` Select your modified RAEM.txt file.

Then `File -> New Contest`, choose RAEM. For the exchange just put in your latlon.

[image]

Then `File -> Configuration Settings`, select the `Options` TAB. and enable the `ESM` and `Use Call History` checkboxes.

[image]

Then give is a spin and let me know if I got close.

mbridak, 2024-11-03:
Oh yeah, your exchange macro should be `# {EXCH}`.

py4xx, 2024-11-08:
Hi Michael, TNX...

I'll test it and tell you later...

py4xx, 2024-11-08:
This is a CallHistory of RAEM to DXLog...

I don't know if it will serve you as an example... 73

# Database for RAEM International Contest
# Data collected and maintained by Claude VE2FK
# Report updates and corrections directly to ***@***.***
# Last updated 2023-12-25
2E0SUR=51N0W
2E0ZSE=51N3W
4K6FO=40N50O
4L8A=41N47O
4O3A=42N18O
4O4SM=43N19O
4Z5KO=32N35O
4Z5LY=32N35O
5B4AMM=35N32O
7J1AAX=35N139O
7K1LUE=35N139O
7Z1HL=25N47O
9A2AJ=45N17O
9A3XV=45N16O
9A4W=43N16O
A61EK=24N54O
A65CA=25N55O
A96A=26N50O
AA3B=40N76W
AA4D=38N82W
AA4DD=37N82W
BA4MY=36N117O
BD1IIJ=41N117O
BG7BDB=29N113O
CR7ACS=40N8W
CT1DRB=38N9W
CT1ELZ=38N9W
CT7/RC2A=37N8W
D1WW=48N37O
D3AA=9S20O
DD0VS=50N11O
DD5M=48N12O
DF1MM=53N09O
DF1SZ=49N9O
DF1XC=52N14O
DF4BV=53N08O
DF4PD=50N7O
DF4TD=47N09O
DF5BX=52N08O
DF6RI=49N11O
DF7AT=53N10O
DF7TT=54N12O
DF8XC=51N8O
DF9EY=51N7O
DG0KS=54N12O
DG6ME=51N10O
DG7EE=50N12O
DG8KAD=51N06O
DH0GHU=49N08O
DH2URF=51N6O
DH3LK=53N9O
DH6YAG=52N8O
DJ1OJ=48N11O
DJ1YF=53N10O
DJ2AX=51N12O
DJ2IA=51N14O
DJ2QV=48N12O
DJ2RG=52N11O
DJ4XD=52N13O
DJ5HB=52N9O
DJ5QV=50N8O
DJ6ZC=53N10O
[... 1434 lines cut ...]
YT1A=43N20O
YT1V=44N19O
YT2AA=44N21O
YT4B=44N19O
YT4W=44N21O
YT6M=43N20O
YT8A=44N20O
YU0T=43N19O
YU1AN=43N19O
YU1FG=44N20O
YU1JW=43N19O
YU1SV=43N22O
YU1TY=44N22O
YU1YV=44N20O
YU2A=43N21O
YU5M=44N22O
YU5R=44N21O
YU7U=45N20O
YU8A=44N21O
Z35F=41N22O
Z35G=42N22O
Z35M=41N21O
ZL1TM=37S175O
ZL3IO=40S177O
ZM4T=40S177O

mbridak, 2024-11-08:
I wrote it to support n1mm call history files. The RAEM one can be found at:

https://n1mmwp.hamdocs.com/mmfiles/raem_2024-txt/

py4xx, 2024-11-09:
Hi Michael... 73.
I tested the CallHistory import here and it worked fine. The only
problem I observed was that if I have a call sign in the call field
and there is an exch in the exch fild, if I delete the call sign
before saving, the exch does not delete. So I have to go to the exch
field and delete it. This takes time and if it is not observed, the
saved callsign will have the wrong exch.
Another thing I noticed is that when I want to use not1mm with FLDigi,
when opening FLDigi, not1mm closes. And it won't open until I close
FLDigi.
Fort 73.

mbridak, 2024-11-09:
You should use the ESC key to clear all fields.
This will clear the both the call and any exchange.

Re: FLDIGI Hmm that's a new one. I'll look at it.

mbridak, 2024-11-09:
Re: FLDigi
[image]

Mine live happily together...

py4xx, 2024-11-10:
> Re: FLDigi
> Screenshot.From.2024-11-09.15-14-34.png (view on web)
>
> Mine live happily together...

Hi Michael... 73
Pse watch my video...

https://drive.google.com/file/d/1fCIDCSNPtZ7kPpU9-jaPX0evXqkp8tMT/view?usp=drive_link

If possible, give me your settings of FLDigi and Not1mm.

 TNX

mbridak, 2024-11-10:
Okay. I noted in your video that your using flrig. So I switched over to use flrig. Then fldigi locked up and became unusable. I wasn't able to change modes.

So I'm going to assume something is hosed with my flrig interface. This would make sense since I hate flrig and try to avoid it whenever possible. I'll put my big boy pants on to work with it more.

I'm going to open a new issue for this, since it's different from the initial problem/request.

mbridak, 2024-11-11:
@py4xx Please see issue: flrig... I hate it. It hates me. #212