mbridak/not1mm #124: contest italian 40/80 ari

ik3ocd, 2024-07-30:
good morning, I'm Flavio ik3ocd, I would like to make a request if possible for your inclusion in the Italian 40/80 contest of the Italian radio amateurs association, thanks for your work to also allow Linux to operate in contests and better graphics. 73

mbridak, 2024-07-30:
Hello Flavio, I can't find anything on this contest. Could you provide a link to a page with the rules/scoring etc?

mbridak, 2024-07-30:
Ah, never mind, I found it. https://www.ari.it/contest-hf/contest-4080.html

Just so you know, this logger currently does not directly support RTTY.

But I'll put this on the TODO list.

ik3ocd, 2024-07-30:
Hi Michael, thanks for your interest, yes I can give you this info in pdf
in Italian or directly from the ari site

keep in mind that this contest has various operating modes, see if it can
be arranged in your not1mm. bye and best regards ik3ocd flavio near venice

https://www.ari.it/contest-hf/contest-4080.html.

ik3ocd, 2024-07-30:
ah ok
Michael ti ringrazio molto per tuo disponibilità 🙌 i miei piu cordiali
saluti e a presto . 73 de ik3ocd flavio

mbridak, 2024-07-31:
added to todo

closed 2024-07-31 by mbridak