mbridak/not1mm #126: N1MM Contact Broadcast does not deliver tx- and rxfrequency

dg9vh, 2024-09-08:
**Describe the bug**
When using the N1MM Data-Export by configuration in settings, the delivered xml does not contain frequencies but only bands.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to configuration settings/N1MM
2. Enable Send N1MM Packets and activate "Contact"
3. Log QSO
4. inspect resulting xml received by Client.

**Expected behavior**
XML should contain the logged qrgs in rx und tx.

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Debian Linux
 - Version 12

mbridak, 2024-09-08:
Thanks for the bug report. I just pushed a fix.

dg9vh, 2024-09-09:
Hi, tnx, will check this this evening - cause is how I found this, I am actually building a service that listens for the logs of N1MM / Not1MM and pushes the QSOs as ADIF-entries into Cloudlog / Wavelog online log.

We used your program this weekend in IARU-R1 Fieldday (by using a different Contest-layout because R1 Fieldday was not in the list)... it worked good so far and I decided, to use it on my Linux-environment as primary contest logger in the future. Therefore I needed this service I develloped but got no QRGs in your data :-)

Will close this issue, if solved. So far,
best 73
Kim

dg9vh, 2024-09-09:
It works as expected :-)

closed 2024-09-09 by dg9vh

mbridak, 2024-09-09:
Excellent. Thanks for checking. And thanks for using it. I'll add the IARU-R1 Fieldday to the todo list.