mbridak/not1mm #135: Upstart complaints

awandahl, 2024-09-14:
**Describe the bug**
aw@5x7w-t2:~$ not1mm
Traceback (most recent call last):
  File "/home/aw/.local/bin/not1mm", line 5, in <module>
    from not1mm.__main__ import run
  File "/home/aw/.local/share/pipx/venvs/not1mm/lib/python3.10/site-packages/not1mm/__main__.py", line 3491, in <module>
    window = MainWindow()
  File "/home/aw/.local/share/pipx/venvs/not1mm/lib/python3.10/site-packages/not1mm/__main__.py", line 474, in __init__
    self.lookup_service = LookupService()
  File "/home/aw/.local/share/pipx/venvs/not1mm/lib/python3.10/site-packages/not1mm/lookupservice.py", line 38, in __init__
    if self.settings.get("useqrz"):
AttributeError: 'NoneType' object has no attribute 'get'

**To Reproduce**
First install trying to follow the instructions

**Desktop (please complete the following information):**

Xfce4

DISTRIB_DESCRIPTION="Ubuntu 22.04.4 LTS"
PRETTY_NAME="Armbian 24.8.2 jammy"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.4 LTS (Jammy Jellyfish)"

**Additional context**
Starting afresh here so need a bit of help.

mbridak, 2024-09-15:
Okay. I pushed a change which should fix this.
Thanks for the report.

awandahl, 2024-09-15:
Works now! Thanks a million! 
73 Anders SM0HPL 5X7W

closed 2024-09-15 by mbridak