mbridak/not1mm #139: Pressing "F2: HisCall" when no callsign in the box makes the winkey server crash

awandahl, 2024-09-17:
**Describe the bug**
Running CW i "Run mode", having the Callsign entry box empty, and then trying to press "F2: HisCall" gives the error [0 recvfrom with UDP] which crashes the WinKey server daemon (https://github.com/ok2cqr/winkeyer_server). I realize I'm on my own when using a third-part winkey application, but it makes no sense trying to send HisCall when there is no HisCall in the box. Restarting winkey server solves the immediate issue.

**Expected behavior**
Nothing should happen if pressing  "F2: HisCall" if box is empty , or at least nothing should be sent to port 6789.

**Desktop (please complete the following information):**
 - OS: Armbian Ubuntu 22.04
 - Version [e.g. 22]

awandahl, 2024-09-17:
Same goes for the buttons named "Empty", nothing should be sent.

mbridak, 2024-09-18:
@awandahl Okay, Don't quite know why the other software is crashing, but to help out. I've pushed a change to master it'll get included in the next published version.

closed 2024-09-18 by mbridak

mbridak, 2024-09-18:
@awandahl 

```
I think the winkeyer application is to blame in this case. 
```

I would agree ;-)