mbridak/not1mm #141: Not1MM closing when pressing SPACE

py4xx, 2024-09-18:
**Describe the bug**
A clear and concise description of what the bug is.

_**With or without call in the callsign field, when you press space, not1mm simply closes.**_

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**_2. Click on SPACE key_**

**Expected behavior**
A clear and concise description of what you expected to happen.

**_Let the cursor go to the next field._**

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] - _**Linux UBUNTU**_
 - Version [e.g. 22]  **_24.04_**

**Additional context**
Add any other context about the problem here.

mbridak, 2024-09-18:
@py4xx Hello, I can't reproduce this. It may be dependent on the contest selected at the time.
Could you tell me which one was used?

mbridak, 2024-09-21:
@py4xx I'll be closing this soon if I don't get a little more info.
I need to know which contest was loaded when this happened.
It would help even more if you launched it from a terminal, then paste in the traceback it spits out when it died.

mbridak, 2024-09-22:
closing for lack of followup.

closed 2024-09-22 by mbridak

py4xx, 2024-09-24:
Hi Michael...

Sorry I didn't respond sooner, but I was really busy... The contest was the
CWT. Whenever I pressed space, the program closed. Fort 73

mbridak, 2024-09-25:
Thanks. Fixed and pushed.