mbridak/not1mm #155: Running dupes give points in the log - ideally shouldn't be so

awandahl, 2024-09-26:
**Is your feature request related to a problem? Please describe.**

Running a station again (a so called dupe) in a contest is not necessarily a bad thing. It's usually convenient not to get into a discussion if "worked B4" is true or not, easier just to run the station again and get rid of the problem. But it's also necessary that you on your side indicates that it is a dupe. This is usually done by putting 0 points in the log.

**Describe the solution you'd like**
Running a dupe should register 0 points in the log for the QSO.

**Describe alternatives you've considered**
Makin a note somewhere and edit by hand afterwards. Not very practical.

awandahl, 2024-10-23:
For CQWW CW, does this make sense?
```
def points(self):
    """Calc point"""
    # Retrieve callsign and band information
    call = self.contact.get("Call", "")
    band = float(get_logged_band(str(self.contact.get("Freq", 0.0))))
    
    # Check if the contact is a duplicate on the current band
    dupe_check = self.database.check_dupe_on_band(call, band)
    if dupe_check.get("isdupe", 0) > 0:
        return 0  # Return zero points for duplicates

    result = self.cty_lookup(self.station.get("Call", ""))
    if result:
        for item in result.items():
            mycountry = item[1].get("entity", "")
            mycontinent = item[1].get("continent", "")
    result = self.cty_lookup(self.contact.get("Call", ""))
    if result:
        for item in result.items():
            entity = item[1].get("entity", "")
            continent = item[1].get("continent", "")
            if mycountry.upper() == entity.upper():
                return 0
            if mycontinent and continent == "NA":
                return 2
            if mycontinent == continent:
                return 1
            return 3
    return 0
```

mbridak, 2024-10-24:
Looks good. I'd merge that.

awandahl, 2024-10-24:
No, it doesn't work. I have problems finding out how to call the database apparently:
```
(not1mm) aw@HP-ZBook:~/exp_not1mm/not1mm-master$ python -m not1mm
Traceback (most recent call last):
  File "/home/aw/exp_not1mm/not1mm-master/not1mm/__main__.py", line 709, in check_esm_with_enter
    self.contest.process_esm(self, with_enter=True)
  File "/home/aw/exp_not1mm/not1mm-master/not1mm/plugins/cq_ww_cw.py", line 461, in process_esm
    self.save_contact()
  File "/home/aw/exp_not1mm/not1mm-master/not1mm/__main__.py", line 2251, in save_contact
    self.contact["Points"] = self.contest.points(self)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aw/exp_not1mm/not1mm-master/not1mm/plugins/cq_ww_cw.py", line 155, in points
    band = float(get_logged_band(str(self.contact.get("Freq", 0.0))))
                 ^^^^^^^^^^^^^^^
NameError: name 'get_logged_band' is not defined
``

awandahl, 2024-10-24:
This beats me! Am I even close??? At least this one doesn't throw any errors but it doesn't work either. And I can't find any log output.
```
def points(self):
    """Calc point"""
    # Retrieve callsign
    call = self.contact.get("Call", "")
    # Use predefined band setting
    band = self.contest_settings.get("BandCategory", "")

    # Check if the contact is a duplicate on the current band
    dupe_check = self.database.check_dupe_on_band(call, band)
    if dupe_check.get("isdupe", 0) > 0:
        return 0  # Return zero points for duplicates

    result = self.cty_lookup(self.station.get("Call", ""))
    if result:
        for item in result.items():
            mycountry = item[1].get("entity", "")
            mycontinent = item[1].get("continent", "")
    result = self.cty_lookup(self.contact.get("Call", ""))
    if result:
        for item in result.items():
            entity = item[1].get("entity", "")
            continent = item[1].get("continent", "")
            if mycountry.upper() == entity.upper():
                return 0
            if mycontinent and continent == "NA":
                return 2
            if mycontinent == continent:
                return 1
            return 3
    return 0
``

mbridak, 2024-10-24:
Nope, you were closer with the first one. The logic is sound. You just have to carefully look at the thrown error.
Fix why the error happened, then if it still doesn't work there must be something wrong with the data.
So you need to look at what the supporting functions need and compare that with the data your feeding them.

So the very first thing you saw was:

```python
NameError: name 'get_logged_band' is not defined
```

So the `points` function doesn't know about the get_logged_band function. To fix that you import it. so you can add:

```python
from not1mm.lib.ham_utility import get_logged_band
```  

near the top where the other imports are. This will stop the crashes, but it still isn't doing what we're asking of it. So the data must be wrong. So we have to look at the data that we're providing to the functions. So just add a little print statement to dump the values we want to see. So after you get the value for dupe_check you can add something like:

```python
print(f"{call=} {band=} {dupe_check=} {self.contact=}")
```

and you'll get:

```text
call='M0MCX' band=0.0 dupe_check={'isdupe': 0} self.contact={'TS': '2024-10-24 14:51:06', 'Call': 'M0MCX', 
'Freq': 14030.7, 'QSXFreq': 14030.7, 'Mode': 'CW', 'ContestName': 'CQ-WW-CW', 'SNT': '599', 'RCV': '599', 
...More stuff truncated for brevity
```

Okay. So we can see that `band` is 0.0, so the problem most likely what we're feeding the `get_logged_band` function. We can look there to see what it expects from us. The doc string for `get_logged_band` says:

```text
    Convert a (string) frequency into a (string) band.
    Returns a (string) band.
    Returns a "0" if frequency is out of band.
```

And a code snippet from it:

```python
if 14350000 > frequency > 14000000:
    return "14"
```

Shows that it's working with frequencies in Hz. But the print statement shows that we are feeding it from `'Freq': 14030.7` which is a float in Khz. We just need to do some data conversion. So:

```python
band = float(get_logged_band(str(self.contact.get("Freq", 0.0))))
```

becomes:

```python
band = float(get_logged_band(str(int(self.contact.get("Freq", 0.0) * 1000))))
``` 

Now we should be feeding the function the correct data and getting some useful work out of the function. Our debug print now gives us:

```python
call='M0MCX' band=14.0 dupe_check={'isdupe': 1}
```

Now the `band` has a good value and the dupe_check is working.

[image]

As we can see above, the first time the qso gives points, and the second does not.

awandahl, 2024-10-24:
Wow! So this is it! Indeed seems to work. Thanks for the pedagogical explanation! I learnt a lot!
```
"""CQ World Wide DX CW plugin"""

# CQ Worldwide DX Contest, CW
#  	Status:	Active
#  	Geographic Focus:	Worldwide
#  	Participation:	Worldwide
#  	Awards:	Worldwide
#  	Mode:	CW
#  	Bands:	160, 80, 40, 20, 15, 10m
#  	Classes:	Single Op All Band (QRP/Low/High)
# Single Op Single Band (QRP/Low/High)
# Single Op Assisted All Band (QRP/Low/High)
# Single Op Assisted Single Band (QRP/Low/High)
# Single Op Overlays: (Classic/Rookie/Youth)
# Multi-Single (Low/High)
# Multi-Two
# Multi-Multi
# Explorer
#  	Max power:	HP: 1500 watts
# LP: 100 watts
# QRP: 5 watts
#  	Exchange:	RST + CQ Zone No.
#  	Work stations:	Once per band
#  	QSO Points:	0 points per QSO with same country (counts as mult)
# 1 point per QSO with different country same continent
# 2 points per QSO with different country same continent (NA)
# 3 points per QSO with different continent
#  	Multipliers:	Each CQ zone once per band
# Each country once per band
#  	Score Calculation:	Total score = total QSO points x total mults
#  	E-mail logs to:	(none)
#  	Upload log at:	https://www.cqww.com/logcheck/
#  	Mail logs to:	(none)
#  	Find rules at:	https://www.cqww.com/rules.htm
#  	Cabrillo name:	CQ-WW-CW

# pylint: disable=invalid-name, unused-argument, unused-variable, c-extension-no-member, unused-import

import datetime
import logging

from pathlib import Path

from PyQt6 import QtWidgets

from not1mm.lib.plugin_common import gen_adif, get_points
from not1mm.lib.version import __version__
from not1mm.lib.ham_utility import get_logged_band

logger = logging.getLogger(__name__)

EXCHANGE_HINT = "CQ Zone No."

name = "CQ WW CW"
cabrillo_name = "CQ-WW-CW"
mode = "CW"  # CW SSB BOTH RTTY
# columns = [0, 1, 2, 3, 4, 5, 6, 15]
columns = [
    "YYYY-MM-DD HH:MM:SS",
    "Call",
    "Freq",
    "Snt",
    "Rcv",
    "SentNr",
    "RcvNr",
    "PTS",
]

[... 381 lines cut ...]
        if self.current_widget == "callsign":
            if len(self.callsign.text()) > 2:
                self.make_button_green(self.esm_dict["MYCALL"])
                buttons_to_send.append(self.esm_dict["MYCALL"])

        elif self.current_widget == "other_2":
            if self.other_2.text() == "":
                self.make_button_green(self.esm_dict["AGN"])
                buttons_to_send.append(self.esm_dict["AGN"])
            elif self.other_2.text().isnumeric():
                self.make_button_green(self.esm_dict["EXCH"])
                buttons_to_send.append(self.esm_dict["EXCH"])
                buttons_to_send.append("LOGIT")
            else:
                self.make_button_green(self.esm_dict["AGN"])
                buttons_to_send.append(self.esm_dict["AGN"])

        if with_enter is True and bool(len(buttons_to_send)):
            for button in buttons_to_send:
                if button:
                    if button == "LOGIT":
                        self.save_contact()
                        continue
                    self.process_function_key(button)
```

mbridak, 2024-10-24:
No problem. Glad to help. and I learned a new word, pedagogical, :+1: 

I've committed the changes and pushed 24-10-24.
I'll close this as complete.

closed 2024-10-24 by mbridak