mbridak/not1mm #166: Change port to 8000 when activating PyWinkeyer

awandahl, 2024-10-08:
**Is your feature request related to a problem? Please describe.**
PyWinkeyer run on port 8000, not 6789 as cwdaemon does. This is not obvious.
**Describe the solution you'd like**
Port numbers change automatically depending of which keying you choose, as a convenience.
**Describe alternatives you've considered**
Clarifying help text.

mbridak, 2024-10-08:
Even tho the port to use is listed in both projects documentation, I pushed a change addressing this issue.

I've made it so the placeholder text for the port numbers change based on which radio button is pressed.

[image]
[image]
[image]

closed 2024-10-08 by mbridak

awandahl, 2024-10-08:
Do people read documentation? ;) No! I did though...