mbridak/not1mm #171: Leading zero in CQWW CW CQ Zone

awandahl, 2024-10-13:
**Describe the bug**
A leading zero is sent in the exchange of CQ Zone, in my case 014

**To Reproduce**
Steps to reproduce the behavior:
1. Run CQ WW CW
2. In this case I'm NOT using ESM feature
3. Exchange in F-key = {SNT} {SENTNR}
4. Same for RUN and S&P

**Expected behavior**
Without leading zero for exchange not being serial number, and even if serial number they could be removed, maybe by a setting...?

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mbridak, 2024-10-14:
@awandahl Hi. I think it would work correctly for you if you changed your exchange macro to `{SNT} {EXCH}`.

awandahl, 2024-10-15:
Of course... thanks!

closed 2024-10-15 by mbridak