mbridak/not1mm #175: Mode macro template is changing when switching from RUN to S&P and back

awandahl, 2024-10-15:
**Describe the bug**
You can close #172 which was a confused way of describing this behavior. I try again here... I haven't seen any obvious pattern for these errors. I'm fake working CQ WW CW in ESM mode and I'm switching between RUN and S&P in a pretty stochastic way. I think these errors are related in some way and they appear both in ESM and non-ESM.

- Mode macro template is *sometimes* changing when switching from RUN to S&P and back, in my case from CW to SSB. I.e. changing from cwmacros.txt to ssbmacros.txt. Can be switched back to cwmacros.txt by entering CW <space> in callsign field.

- Report type changes from 5NN to 599 for a non obvious reason. Can be reset back to 5NN by entering SSB <space> CW <space> in callsign field.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

awandahl, 2024-10-16:
Maybe anecdotal, but changing between SSB and CW by typing in the callsign field followed by space bar is working kind of flawless if NOT in ESM mode, while it's easy to change to SSB but a hassle to get it back to CW, several tries needed usually, when IN ESM mode.

mbridak, 2024-10-17:
I've had time finally to visit the shack and test it out on a live radio. It seems when switching modes by typing "CW" or "SSB" back and forth repeatedly that the CAT link (I've only tested rigctld) sometimes fails. When  the link comes back up, the app will sync back to what mode the radio is in. Not sure what is causing this. I'll need to noodle on it a bit.

mbridak, 2024-10-17:
Upgrade to 24-10-17-1 and let me know if it is any better.

awandahl, 2024-10-17:
MUCH better! Thanks! But it still happens sometimes. I run a Elecraft K2 which is at 4800 bd. Will try my QRP Labs QMX too. I also use rigctld btw. I appreciate your agility immensely! CQWW CW with not1mm in Nov seems like a possibility.

awandahl, 2024-10-18:
Good news! Now I have tried with the QMX and it's working flawless. This can be the reason  ```stty -F /dev/ttyACM0 speed``` = 115200 while ```stty -F /dev/ttyACM0 speed``` = 4800. The Elecraft K2 use old RS232. The changes you did yesterday improved the K2 performance a lot though!

mbridak, 2024-10-19:
@awandahl Okay, I rewrote parts of the CAT control and how the interface is updated.

It used to push the change to the radio and then manually change the interface to match so it would look snappier. But sometimes the thread that polls the radio would get called in the middle of the write to the radio. which would revert the interface back when it was next polled

Rewrote it a bit to just push the change to the radio, and then wait for the poll to normally happen to sync up the state.
So hopefully it won't jump back and forth now. 

So update to 24-10-19-1 and let me know if it's "Better".

awandahl, 2024-10-24:
It's *much better* now! I have dry run more than 1000 QSOs now without any signs of mode changing or any other flaws.
CQWW CW is coming closer... I'm quite confident that it will work out fine!

mbridak, 2024-10-24:
Cool. I'll close this as complete.

closed 2024-10-24 by mbridak