mbridak/not1mm #176: Show red text "Dupe" already when call is in callsign field

awandahl, 2024-10-15:
**Is your feature request related to a problem? Please describe.**
The dupe is indicated *after* QSO has started and when reading/entering other station's exchange, which is too late.

**Describe the solution you'd like**
The red "Dupe" could be shown already when the call is in the callsign field. You then have to decide if you want to work the station (to get rid of hassle) or say QSO B4. It would also be extremely comfortable when using up/down arrows for navigating the bandmap, easier to see the dupes when the map is crowded.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mbridak, 2024-10-17:
Fixed w/ 24-10-17-1.

closed 2024-10-17 by mbridak

awandahl, 2024-10-17:
Simply perfect!