mbridak/not1mm #179: Hotkeys can be assigned according to what N1MM use

awandahl, 2024-10-16:
Hotkeys should preferably be the same as in the mothership, to make a shift from one to the other swift and painless. I will work on this ticket a while and use it as a notepad for my thinking and reasoning.

[CTRL-M] Mark Callsign to the bandmap window to work later.

Now: [CTRL-S]	Spot Callsign to the cluster
"""
Alt+P – Spot the callsign of the last station logged or the one currently in the callsign textbox in the format DX CALL Frequency, without comment unless one has been specified on the Spot Comment tab of the Telnet window.
Ctrl+P – Spot the callsign of the last station logged or the one currently in the callsign textbox. You will be prompted for a comment. Hit ESC,click “Cancel” or click the red X to exit the comment window without spotting. If you do not enter a comment, and one has been specified on the Spot Comment tab, it will be sent when you click OK. If you specify a comment, it will be sent along with the one specified on the Spot Comment tab.
"""

py4xx, 2024-11-14:
1.384 / 5.000
Hello...
Today I participated in the CWT with our Not1MM and I confess that I had a lot of difficulty controlling it.
The shortcut keys, which are different from those used in other contest programs, really bothered me.
To delete fields, in Not1mm, we use ESC, but in other programs ESC is to interrupt the transmission. This caused me some errors, because I pressed ESC to interrupt the TX and it deleted everything... LOL
The suggestion would be a way for the user to configure the shortcut keys for the best control for him/her.
For example, I use the CTRL + W keys to delete the CALL and EXCH fields. So I would replace ESC with them and set it to interrupt TX.

Another thing I was missing was CQ AUTO.

Another thing that would be ideal for me would be that when I typed CALL, the callhistory information would appear, without having to press space and go back to the CALLSIGN field to call or answer.
I would like to be able to increase the font of the Check Parts List... For me, who has vision problems, it is too small and with the green strip with white letters in the Master, it doesn't help much. (I use DARK mode)

[image]

The implementation of CallHistory was excellent.

I believe that with these changes, we will be fulfilling a dream of us amateur radio competitions, of having a native Linux program (without Wine) as competitive as the others for Windows... 
We are on the right track.

Congratulations...

mbridak, 2024-11-14:
Everyone wants key bindings that they are used to. I get emails from people who want TLF key bindings. They use CTRL-A to mark a spot on the band map and CTRL-B to spot to the cluster, and ESC stops the CW but also wipes the inputs at the same time. 
As far as the control key assignments go, A lot of the key combinations are consumed by the input fields themselves for editing functionality. Like CTRL-U clears the current field. Did a quick check and these are the unconsumed CTRL key combinations.

`CTRL + QWRTYIOPSFGHJLBNM` are free combinations. 

I'm torn on the ESC key. Currently, ESC clears inputs and CTRL-ESC stops transmission. I can see changing this. Stopping the transmission is more of a panic stop situation. Say you were S&P and inadvertently responded thinking the running station was replying to you, but you're now in fact QRMing and you want to stop ASAP. It would be easier for your lizard brain to slap the ESC key to stop than to CTRL-ESC. And I can see changing the clear fields to CTRL-W, since in your head you can think Wipe.    

The 'Adult' thing to do would be to see if I can override the input class and steal all the keys. Then make an interface for people to define their own key bindings. The first part seems fun. The second... not so much. But I'll squirrel it away in the todo file.

Auto CQ:
I've been thinking about this. The only reason it's not already done is I'm a 'little pistol' without the confidence in his CW skills to RUN. But I've been thinking of forcing myself to try since if the other side really wants the points they will suffer thru my agn agn agn... Maybe I'll make the first Auto AGN feature. Since I'm a little pistol I'll need the Auto CQ.

>>>Another thing that would be ideal for me would be that when I typed CALL, the callhistory information would appear, without having to press space and go back to the CALLSIGN field to call or answer.

I'm not following this and would need clarification.

Font size in check partial window: I can do that.

Colors in check partial window:

I'd need an example color combination, preferably with hex or decimal RGB values. I myself am color blind and am the worst judge. I also really couldn't make out the matched call, but didn't care 'cause I knew if the background was green, I knew it was a match to what was in the callsign field.
Also I rarely use this window, since my laptop is ancient and this widget/window is a resource hog. It slows the main interface down so much I find it troublesome and prefer to do without.

py4xx, 2024-11-15:
Hi Michael... 73...

 ** I'm not following this and would need clarification.

 I'll try to explain better...
What I noticed is that when I type the other colleague's callsign in the callsign field, the CallHistory information doesn't appear...
Ex.: I type the callsign K3WW in the callsign field and the name and number don't appear in the exch field. I have to press SPACE to display the information. My suggestion is that this information be displayed without the need to press space.
If I didn't make myself clear again, please let me know and I'll try to make a video. 73.

mbridak, 2024-11-15:
No, I don't think that would work. I think it is important that the call history data not overwrite information already existing in the exchange fields. So say you're typing in a call and that call is AA0ZZ, as you're typing it will first find AA0Z. The exchange info will populate for AA0Z, then you key the last Z. It will match AA0ZZ but the exchange info will still be for AA0Z since it already exists in the exhange field.

mbridak, 2024-11-26:
If you're using full breakin you got about an equal chance to get a false rx state.

I'm going to change it to ESC - stop CW and CTRL-W for wipe.

awandahl, 2024-11-26:
"If you're using full breakin you got about an equal chance to get a false rx state." Maybe... but is it possible to try out? I'm kind in love with the idea. If it works it will be a function one step ahead of DXlog and N1MM. I liked to wipe with ESC, that was also natural... Otherwise a move to the "normal" behaviour of N1MM is fine with me. Thanks!

closed 2025-02-23 by mbridak