mbridak/not1mm #193: Program crash when exporting to Cabrillo

awandahl, 2024-10-24:
**Describe the bug**
Program crash when exporting to Cabrillo. I have tried CQWW CW and WPX CW, same result. Nothing to be seen in not1mm_debug.log  The file is created in home directory with preamble, but no QSOs
```
 START-OF-LOG: 3.0
CREATED-BY: Not1MM v24.10.23
CONTEST: CQ-WW-CW
CLUB: GQRP
CALLSIGN: SE0I
LOCATION: DX
CATEGORY-OPERATOR: SINGLE-OP
CATEGORY-ASSISTED: NON-ASSISTED
CATEGORY-BAND: ALL
CATEGORY-MODE: CW
CATEGORY-TRANSMITTER: ONE
GRID-LOCATOR: JO89XI
CATEGORY-POWER: QRP
CLAIMED-SCORE: 340561
OPERATORS: @SE0I, SE0I
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mbridak, 2024-10-24:
That's odd. Mine's working fine.

```text
START-OF-LOG: 3.0
CREATED-BY: Not1MM v24.10.24
CONTEST: CQ-WW-CW
CALLSIGN: K6GTE
LOCATION: ORG
CATEGORY-OPERATOR: SINGLE-OP
CATEGORY-ASSISTED: ASSISTED
CATEGORY-BAND: ALL
CATEGORY-MODE: CW
CATEGORY-TRANSMITTER: ONE
GRID-LOCATOR: dm13at
CATEGORY-POWER: LOW
CLAIMED-SCORE: 18
OPERATORS: @K6GTE, K6GTE
NAME: Michael Bridak
ADDRESS: 2854 W Bridgeport Ave
ADDRESS-CITY: Anaheim
ADDRESS-STATE-PROVINCE: CA
ADDRESS-POSTALCODE: 92804
ADDRESS-COUNTRY: United States
EMAIL: michael.bridak@gmail.com
QSO: 14030 CW 2024-10-24 1548 K6GTE         599 3      M0MCX         599 14    
QSO: 14030 CW 2024-10-24 2129 K6GTE         599 3      SE0I          599 14    
END-OF-LOG:
```

Your's is stopping where the it would output your name, That's an odd place to stop. I can't think or much that would cause that. Maybe a character encoding problem?

I checked your name on QRZ The problem is probably with the Ä in your last name.

I'll test it with your name.

mbridak, 2024-10-24:
Okay, I tried it with your name and got:
```text
Traceback (most recent call last):
  File "/home/mbridak/Nextcloud/dev/not1mm/not1mm/__main__.py", line 3646, in generate_cabrillo
    self.contest.cabrillo(self)
  File "/home/mbridak/Nextcloud/dev/not1mm/not1mm/plugins/cq_ww_cw.py", line 327, in cabrillo
    print(
UnicodeEncodeError: 'ascii' codec can't encode character '\xe4' in position 14: ordinal not in range(128)
```
This will take me some noodling to work out. This could effect quite a few things like the address or club names.
Speaking of which, It's always a good day when the SPRAT journal shows up in the mail.

I'll work on it.

mbridak, 2024-10-24:
When I get back home I'll try opening the file in UTF-8 mode. That will probably clear it up. Don't know if all the contest administrators software handles UTF-8 files. You would hope so.

mbridak, 2024-10-25:
I've pushed 24-10-24-1 which fixes this.

mbridak, 2024-10-25:
Maybe, I sent the a message to the CQ WW / CQ WPX group thru their contact form asking if they accept UTF-8.

If they don't get back to me soon I'll probably change at-least the WW DX SSB one back to ASCII to be safe.

awandahl, 2024-10-25:
I saw this old issue in the discussion btw https://github.com/mbridak/not1mm/discussions/64  Exactly the same thing I gather...

mbridak, 2024-10-25:
You're probably right. Too bad they didn't follow up.

awandahl, 2024-10-25:
DXLog have these commands, adding an "a" for pure ASCII. Could be a way to handle things in not1mm too...
```
WRITELOG or MAKELOG 	Exports the log to a file in Cabrillo, EDI, or ADIF format depending of type of contest.
WRITELOGA or MAKELOGA 	Exports the log to a file in Cabrillo, EDI, or ADIF format depending of type of contest.
Uses plain ASCII and does not support international characters. 
``

mbridak, 2024-10-25:
Hmm, Kinda like that.
I think I'll go that route.

awandahl, 2024-10-25:
Haha... I'm pretty used to the *ignore* alternative actually. This is how it usually looks,  fine with me.

mbridak, 2024-10-25:
Okay, We now have a file menu option for the user to choose between ascii that doesn't crash or utf8 that doesn't crash.

closed 2024-10-26 by mbridak