mbridak/not1mm #195: Windows-Arrangement does not stay as it was configured after restart

dg9vh, 2024-10-25:
**Describe the bug**
The Log-area and the previous QSOs with the same callsign arrangement does not stay as been configured after restart.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Not1mm
2. resize log-window and history
3. restart not1mm
4. see the problem
**Expected behavior**
The window should stay as it was configured in last run.

**Screenshots**
before restart:
[image]

after restart:
[image]

**Desktop (please complete the following information):**
 - OS: Debian Linux 12

mbridak, 2024-10-25:
Hmm, That's ugly. Okay I'll look at it after I finish the Cabrillo ASCII/UTF-8 problem I'm working on.

dg9vh, 2024-10-25:
No problem - the solution is to rearrange it after restart ;-) Cabrillo is higher priority :-)

mbridak, 2024-10-27:
https://github.com/user-attachments/assets/f66c2b72-3f5d-4052-9ab6-91254185f1f2

I tried with both X and Wayland and I can't get it to not restore window sizes.

mbridak, 2024-10-27:
I also installed Debian 12 into a VM. Ran it under a Gnome X11 session.
I'm able to detach all the dock widgets from the main window, resize and reposition them all and they get restored without issue.

Also tried it on a Manjaro KDE X11 session and get the same results.

Sorry, not sure why you're haveing trouble.

awandahl, 2024-10-27:
I get the same problem here with xubuntu 24.04. So there is something that needs to be narrowed down. Xfce desktop on my side if that matters.

mbridak, 2024-10-28:
Dunno, May it happens to be the version of the installed Qt and PyQt libraries. We're all using the same code base. So the only thing different is the code base running it...

So far I've tried the following.

Manjaro KDE 
Plasma 6.1.5
Wayland and X11

Python 3.12.7 (main, Oct  1 2024, 00:00:00) [GCC 14.2.1 20240912 (Red Hat 14.2.1-3)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyQt6.QtCore import QT_VERSION_STR, PYQT_VERSION_STR
>>> print(f"{QT_VERSION_STR=} {PYQT_VERSION_STR=}")
QT_VERSION_STR='6.7.1' PYQT_VERSION_STR='6.7.1'

---

Fedora 40, GNOME 46.6 / KDE Plasma 6
Wayland and X11

Python 3.12.7 (main, Oct  1 2024, 00:00:00) [GCC 14.2.1 20240912 (Red Hat 14.2.1-3)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyQt6.QtCore import QT_VERSION_STR, PYQT_VERSION_STR
>>> print(f"{QT_VERSION_STR=} {PYQT_VERSION_STR=}")
QT_VERSION_STR='6.7.1' PYQT_VERSION_STR='6.7.1'

---

Debian 12 GNOME 43.9
Wayland and X11

Python 3.11.2 (main, May  2 2024, 11:59:08) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyQt6.QtCore import QT_VERSION_STR, PYQT_VERSION_STR
>>> print(f"{QT_VERSION_STR=} {PYQT_VERSION_STR=}")
QT_VERSION_STR='6.7.1' PYQT_VERSION_STR='6.7.1'

---

I'm downloading xubuntu 24.04 now.

mbridak, 2024-10-28:
Okay, just installed fresh xubuntu 24.04 from the iso. 
Installed python3-pip, pipx, libxcb-cursor0 python3-pyqt6
installed not1mm with pipx.

got windows to undock (X11) resize and repositioned all. quit main app. and restarted. everything came up where they were before and at the size they were when the app was quit.

So Still not sure whats up...

dg9vh, 2024-10-28:
If you tell me, what to do, I will delivere you as much Information you need.

awandahl, 2024-10-28:
``
(not1mm) aw@HP-ZBook:~/exp_not1mm$ python check_env.py 
--- Debug Information ---
Qt version: 6.7.1
PyQt version: 6.7.1
Python version: 3.12.3 (main, Sep 11 2024, 14:17:37) [GCC 13.2.0]
Operating System: Linux 6.8.0-47-generic
Machine: x86_64
PIPX_HOME: None
PIPX_BIN_DIR: None
Qt Prefix Path: /home/aw/exp_not1mm/not1mm/lib/python3.12/site-packages/PyQt6/Qt6
Qt Library Path: /home/aw/exp_not1mm/not1mm/lib/python3.12/site-packages/PyQt6/Qt6/lib
Qt Plugin Path: /home/aw/exp_not1mm/not1mm/lib/python3.12/site-packages/PyQt6/Qt6/plugins
--- Environment Variables ---
SHELL: /bin/bash
SESSION_MANAGER: local/HP-ZBook:@/tmp/.ICE-unix/2594,unix/HP-ZBook:/tmp/.ICE-unix/2594
WINDOWID: 73400323
QT_ACCESSIBILITY: 1
COLORTERM: truecolor
XDG_CONFIG_DIRS: /etc/xdg/xdg-xubuntu:/etc/xdg
XDG_SESSION_PATH: /org/freedesktop/DisplayManager/Session0
XDG_MENU_PREFIX: xfce-
GTK_IM_MODULE: ibus
CLUTTER_BACKEND: x11
GNOME_KEYRING_CONTROL: /run/user/1000/keyring
LANGUAGE: en_US
SSH_AUTH_SOCK: /run/user/1000/keyring/ssh
XMODIFIERS: @im=ibus
DESKTOP_SESSION: xubuntu
XDG_SEAT: seat0
PWD: /home/aw/exp_not1mm
LOGNAME: aw
XDG_SESSION_DESKTOP: xubuntu
QT_QPA_PLATFORMTHEME: gtk2
XDG_SESSION_TYPE: x11
PANEL_GDK_CORE_DEVICE_EVENTS: 0
GPG_AGENT_INFO: /run/user/1000/gnupg/S.gpg-agent:0:1
XAUTHORITY: /home/aw/.Xauthority
XDG_GREETER_DATA_DIR: /var/lib/lightdm-data/aw
GDM_LANG: en_US
HOME: /home/aw
LANG: en_US.UTF-8
LS_COLORS: rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.crdownload=00;90:*.dpkg-dist=00;90:*.dpkg-new=00;90:*.dpkg-old=00;90:*.dpkg-tmp=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:*.swp=00;90:*.tmp=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:
XDG_CURRENT_DESKTOP: XFCE
VIRTUAL_ENV: /home/aw/exp_not1mm/not1mm
VTE_VERSION: 7600
XDG_SEAT_PATH: /org/freedesktop/DisplayManager/Seat0
CLUTTER_IM_MODULE: ibus
LESSCLOSE: /usr/bin/lesspipe %s %s
XDG_SESSION_CLASS: user
TERM: xterm-256color
GTK_OVERLAY_SCROLLING: 0
LESSOPEN: | /usr/bin/lesspipe %s
USER: aw
DISPLAY: :0.0
SHLVL: 1
QT_IM_MODULE: ibus
XDG_VTNR: 7
XDG_SESSION_ID: c2
VIRTUAL_ENV_PROMPT: (not1mm) 
XDG_RUNTIME_DIR: /run/user/1000
PS1: (not1mm) \[\e]0;\u@\h: \w\a\]${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ 
DEBUGINFOD_URLS: https://debuginfod.ubuntu.com 
XDG_DATA_DIRS: /usr/share/xubuntu:/usr/share/xfce4:/usr/local/share:/usr/share:/var/lib/snapd/desktop:/usr/share
PATH: /home/aw/exp_not1mm/not1mm/bin:/home/aw/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/aw/.local/bin:/home/aw/.local/bin
GDMSESSION: xubuntu
DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus
OLDPWD: /home/aw/exp_not1mm/not1mm
_: /home/aw/exp_not1mm/not1mm/bin/python
`

awandahl, 2024-10-28:
I used this code to get the info above. And it's from a venv i created locally...
```
def print_debug_info():
    from PyQt6.QtCore import QT_VERSION_STR, PYQT_VERSION_STR, QLibraryInfo
    import sys
    import os
    import platform

    print("--- Debug Information ---")
    print(f"Qt version: {QT_VERSION_STR}")
    print(f"PyQt version: {PYQT_VERSION_STR}")
    print(f"Python version: {sys.version}")
    print(f"Operating System: {platform.system()} {platform.release()}")
    print(f"Machine: {platform.machine()}")
    print(f"PIPX_HOME: {os.environ.get('PIPX_HOME')}")
    print(f"PIPX_BIN_DIR: {os.environ.get('PIPX_BIN_DIR')}")
    print(f"Qt Prefix Path: {QLibraryInfo.path(QLibraryInfo.LibraryPath.PrefixPath)}")
    print(f"Qt Library Path: {QLibraryInfo.path(QLibraryInfo.LibraryPath.LibrariesPath)}")
    print(f"Qt Plugin Path: {QLibraryInfo.path(QLibraryInfo.LibraryPath.PluginsPath)}")
    print("--- Environment Variables ---")
    for key, value in os.environ.items():
        print(f"{key}: {value}")
    print("---------------------------")

print_debug_info()
´´

mbridak, 2024-10-28:
I didn't see anything that jumped out at me. I had hopes it was going to be the Qt and PyQt6 versions, but we're all using 6.7.1 or better.  Then I hoped for X11/Wayland. But we're all using X11. And even Wayland lets you resize the main and dockwidget windows, it just doesn't let you detach the widgets and move them. and it always launches the windows in the middle of the screen. While X11 allows you to restore to a specific location for each widget. Anyway I just don't know... I've included my debug info from the script above for completeness.

```text
--- Debug Information ---
Qt version: 6.7.1
PyQt version: 6.7.1
Python version: 3.12.7 (main, Oct  1 2024, 00:00:00) [GCC 14.2.1 20240912 (Red Hat 14.2.1-3)]
Operating System: Linux 6.11.4-201.fc40.x86_64
Machine: x86_64
PIPX_HOME: None
PIPX_BIN_DIR: None
Qt Prefix Path: /home/mbridak/.local/lib/python3.12/site-packages/PyQt6/Qt6
Qt Library Path: /home/mbridak/.local/lib/python3.12/site-packages/PyQt6/Qt6/lib
Qt Plugin Path: /home/mbridak/.local/lib/python3.12/site-packages/PyQt6/Qt6/plugins
--- Environment Variables ---
STEAM_FRAME_FORCE_CLOSE: 1
GNOME_SETUP_DISPLAY: :1
LMOD_PKG: /usr/share/lmod/lmod
GDMSESSION: gnome
SDL_VIDEO_MINIMIZE_ON_FOCUS_LOSS: 0
DESKTOP_SESSION: gnome
PWD: /home/mbridak
WAYLAND_DISPLAY: wayland-0
LOGNAME: mbridak
LMOD_VERSION: 8.7.49
EDITOR: /usr/bin/nano
MAIL: /var/spool/mail/mbridak
XDG_SESSION_TYPE: wayland
SYSTEMD_EXEC_PID: 2126
PATH: /home/mbridak/bin:/home/mbridak/.local/bin:/usr/local/bin:/usr/lib64/ccache:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin
SSH_ASKPASS: /usr/libexec/openssh/gnome-ssh-askpass
HOME: /home/mbridak
USERNAME: mbridak
QT_IM_MODULE: ibus
HISTCONTROL: ignoredups
LMOD_CMD: /usr/share/lmod/lmod/libexec/lmod
USER: mbridak
DISPLAY: :0
LANG: en_US.UTF-8
LMOD_DIR: /usr/share/lmod/lmod/libexec
XAUTHORITY: /run/user/1000/.mutter-Xwaylandauth.OM8WV2
HOSTNAME: fredo
SESSION_MANAGER: local/unix:@/tmp/.ICE-unix/2081,unix/unix:/tmp/.ICE-unix/2081
XDG_RUNTIME_DIR: /run/user/1000
XDG_SESSION_DESKTOP: gnome
XDG_DATA_DIRS: /home/mbridak/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/
VTE_VERSION: 7604
XDG_CURRENT_DESKTOP: GNOME
MODULEPATH_ROOT: /usr/share/modulefiles
MOZ_GMP_PATH: /usr/lib64/mozilla/plugins/gmp-gmpopenh264/system-installed
LESSOPEN: ||/usr/bin/lesspipe.sh %s
MODULEPATH: /etc/modulefiles:/usr/share/modulefiles:/usr/share/modulefiles/Linux:/usr/share/modulefiles/Core:/usr/share/lmod/lmod/modulefiles/Core
GNOME_TERMINAL_SERVICE: :1.546
MEMORY_PRESSURE_WRITE: c29tZSAyMDAwMDAgMjAwMDAwMAA=
GDM_LANG: en_US.UTF-8
SHELL: /usr/bin/zsh
LMOD_sys: Linux
HISTSIZE: 50000
XDG_SESSION_CLASS: user
XMODIFIERS: @im=ibus
DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus
MEMORY_PRESSURE_WATCH: /sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/session.slice/org.gnome.Shell@wayland.service/memory.pressure
FPATH: /home/mbridak/.oh-my-zsh/plugins/git:/home/mbridak/.oh-my-zsh/functions:/home/mbridak/.oh-my-zsh/completions:/home/mbridak/.oh-my-zsh/custom/functions:/home/mbridak/.oh-my-zsh/custom/completions:/home/mbridak/.oh-my-zsh/cache/completions:/usr/local/share/zsh/site-functions:/usr/share/zsh/site-functions:/usr/share/zsh/5.9/functions:/usr/share/lmod/lmod/init/ksh_funcs
OLDPWD: /home/mbridak
GNOME_TERMINAL_SCREEN: /org/gnome/Terminal/screen/755aefbf_8bf2_47da_958e_ad7bef9831f8
LMOD_SETTARG_FULL_SUPPORT: no
DEBUGINFOD_URLS: https://debuginfod.fedoraproject.org/ 
TERM: xterm-256color
XDG_MENU_PREFIX: gnome-
MODULESHOME: /usr/share/lmod/lmod
LMOD_ROOT: /usr/share/lmod
KDEDIRS: /usr
MANPATH: /usr/share/lmod/lmod/share/man:
COLORTERM: truecolor
SSH_AUTH_SOCK: /run/user/1000/keyring/ssh
BASH_ENV: /usr/share/lmod/lmod/init/bash
SHLVL: 1
LS_COLORS: ***Removed for brevity***
ZSH: /home/mbridak/.oh-my-zsh
PAGER: less
LESS: -R
_: /usr/bin/python
---------------------------
```

awandahl, 2024-10-28:
Obviously the window state and geometry is saved in .config  I can find a directory named K6GTE with some info in a file called not1mm.conf  I tried to copy that directory and renamed it with my contest call SE0I and hoped that would do the trick since that callsign is in may station settings, but alas... I would also have expected that a new directory should be created in that directory when a new callsign/station setting is added. Don't know if I'm on the right track here...?

mbridak, 2024-10-29:
No the QSettings feature is pretty straight forward.

You create a settings object by passing in a company name and a program name.

```python
self.settings = QSettings("K6GTE", "not1mm")
```
the Qt library stores it in ~/.config/company_name/program_name.conf

Near the start of the program it checks if there are key names present to restore from.

```python
if self.settings.value("windowState") is not None:
    self.restoreState(self.settings.value("windowState"))
if self.settings.value("geometry") is not None:
    self.restoreGeometry(self.settings.value("geometry"))
```

When there is a window close event the values are stored and sync() writes it to disk.

```python
self.settings.setValue("geometry", self.saveGeometry())
self.settings.setValue("windowState", self.saveState())
self.settings.sync()
```

And thats about it...

mbridak, 2024-10-29:
If you were to run this:

```python
from PyQt6.QtCore import QSettings
a=QSettings("K6GTE", "not1mm")
print(f"Stored Keys {a.allKeys()}\n\n")
print(f"{a.value('geometry')=}\n\n{a.value('windowState')=}")
```

do you get anything like:

```textStored Keys ['geometry', 'windowState']

a.value('geometry')=PyQt6.QtCore.QByteArray(b'\x01\xd9\xd0\xcb\x00\x03\x00\x00\xff\xff\xff\xfd\xff\xff\xff\xe2\x00\x00\x05G\x00\x00\x02\xd6\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05D\x00\x00\x02\xd3\x00\x00\x00\x00\x00\x00\x00\x00\x06@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05D\x00\x00\x02\xd3')

a.value('windowState')=PyQt6.QtCore.QByteArray(b'\x00\x00\x00\xff\x00\x00\x00\x00\xfd\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x01-\x00\x00\x02\xbb\xfc\x02\x00\x00\x00\x03\xfb\x00\x00\x00\x1c\x00b\x00a\x00n\x00d\x00m\x00a\x00p\x00-\x00w\x00i\x00n\x00d\x00o\x00w\x01\x00\x00\x00\x19\x00\x00\x02\xbb\x00\x00\x00\xcf\x00\xff\xff\xff\xfb\x00\x00\x00\x1a\x00B\x00a\x00n\x00d\x00M\x00a\x00p\x00W\x00i\x00n\x00d\x00o\x00w\x01\x00\x00\x00\x19\x00\x00\x00\xeb\x00\x00\x00\x00\x00\x00\x00\x00\xfb\x00\x00\x00$\x00C\x00h\x00e\x00c\x00k\x00P\x00a\x00r\x00t\x00i\x00a\x00l\x00W\x00i\x00n\x00d\x00o\x00w\x01\x00\x00\x01\n\x00\x00\x00\xae\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\xe2\x00\x00\x02\xbb\xfc\x02\x00\x00\x00\x02\xfb\x00\x00\x00\x18\x00c\x00h\x00e\x00c\x00k\x00-\x00w\x00i\x00n\x00d\x00o\x00w\x01\x00\x00\x00\x19\x00\x00\x02\xbb\x00\x00\x00x\x00\xff\xff\xff\xfb\x00\x00\x00\x14\x00v\x00f\x00o\x00-\x00w\x00i\x00n\x00d\x00o\x00w\x02\x00\x00\x04\x8e\x00\x00\x011\x00\x00\x00\xe2\x00\x00\x00[\x00\x00\x00\x02\x00\x00\x03*\x00\x00\x01\xc2\xfc\x01\x00\x00\x00\x02\xfb\x00\x00\x00\x14\x00l\x00o\x00g\x00-\x00w\x00i\x00n\x00d\x00o\x00w\x01\x00\x00\x013\x00\x00\x03*\x00\x00\x00F\x00\xff\xff\xff\xfb\x00\x00\x00\x14\x00D\x00o\x00c\x00k\x00W\x00i\x00d\x00g\x00e\x00t\x01\x00\x00\x00\x00\x00\x00\x02\xc9\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03*\x00\x00\x00\xf3\x00\x00\x00\x01\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x02\xfc\x00\x00\x00\x00')
```

Mind you I don't know what all the hex values exactly mean, but do you get anything like it?

dg9vh, 2024-10-29:
Hi, on my side, I get similar output. Actual version of not1mm restarts with usable layout of 1/3 log, 1/3 call history and 1/3 entry area, which is far better than my screenshot above. But when resizing entry area to have it a bit smaller, this is not stored.

awandahl, 2024-10-29:
Yes, a lot of interesting hex here too...

mbridak, 2024-11-11:
@awandahl @dg9vh  Okay, new development. I had an issue where after I started using KDE Plasma as my DE, I noticed that the application wasn't displaying as a native KDE app. While researching this I found that if I used pyqt6 installed with pip the app would not appear as a native app, but the windows would restore like it should. But if I removed the pip installed pyqt6 and pyqt6-sip and then reinstalled the python3-pyqt6, python3-pyqt6-base and python3-pyqt6-sip distro packages it would look like a native app but would no longer correctly restore window position and sizes. Also looks like they fixed the detachable widgets in wayland with pyqt version 6.8

So it's nothing I can fix on my side. It all has to do with the how the pyqt library was installed and it's version.

dg9vh, 2024-11-11:
Well, I have distro-packages of Debian 12 since beginning but I can live with the actual layout... it's much better than that I was starting weeks ago :-)

closed 2024-11-12 by mbridak