mbridak/not1mm #198: Zone number stuck after changing call editing for CQ WW

mcogoni, 2024-10-26:
Hi,
if you enter the wrong call and you edit it, the zone number sticks to the first wrong one.
The zone gets recomputed only when resetting with ESC.

Thanks for the nice logging program!

73,
marco IS0KYB

mbridak, 2024-10-26:
Hi the zone number only auto populates if the field is empty. 
The thought behind this is that the human should know more than the computer.
And if there is a value there already it's not going to know if you meant to put it there.
Say if someone was visiting from a zone 3 to a 4 or 5 while operating you'd want to override what's the computer thinks.

What does need to change is if your using the up/down arrows to go to the next call in the bandmap, the Call of the tuned spot changes but the zone stays the same as the previous.

So then the bandmap sends a TUNE signal with the new call and freq, the program should clear the input fields before tuning ad repopulating the call field so the correct zone populates.

mcogoni, 2024-10-26:
I agree that if I enter something in the zone field the program should do nothing, but if I type a wrong call, the zone is guessed and then I change the call back, the zone must be reset too. Don't you agree? ;-)

mbridak, 2024-10-26:
Sure, but it's a state machine type problem. the program doesn't know why there is data in the field.
It just knows there is data in the field...

mbridak, 2024-10-27:
I've been noodling on it a bit, and I think maybe the right solution would be to not auto populate the CQ Zone field until the user leaves the callsign edit field. That way you can edit all you want before it's populated.
I'll work on it a bit.

mcogoni, 2024-10-27:
Yeah, it's a good idea, just a tab more but more control...

mbridak, 2024-10-28:
Better late than never, I pushed 24-10-27.2 which should resolve this.

closed 2024-10-28 by mbridak

mcogoni, 2024-10-28:
Anyway, I used your log during this weekend CQ WW, I did roughly 900 contacts and it was incredibly transparent in its use! VERY GOOD!

mbridak, 2024-10-28:
Excellent! Very good. I'm glad you were able to use it well. Thanks for your input in making it better.
Cheers, Mike K6GTE