mbridak/not1mm #203: Grid location doesn't change with new entery

Bruce-N6BMW, 2024-10-27:
**Describe the bug**
When making new entry Lookup works but the Grid field is not updating.

**To Reproduce**
Steps to reproduce the behavior:
1. Enter callsign
2. Hit 'Enter'
3. Right-click on last entry 
4. See Grid field not change from previous entry; sometimes back a few.

**Expected behavior**
Grid field updated with correct data as per Lookup

**Screenshots**
n/a
**Desktop (please complete the following information):**
 - OS: Ubuntu 
 - Version: 22.04 LTS

**Additional context**
n/a.

mbridak, 2024-10-27:
Hi, Which lookup service are you using?
And just in case what is the call you typed in that the lookup failed with?

mbridak, 2024-10-27:
Ah, I think I think I see what is going on. From your description you enter the call then press enter.
The call isn't looked up until you exit the call field with a SPACE or TAB.
If you just enter the call and press Enter, no lookup has taken place so there is nothing to store.

And Ding! the light just went on.

Okay, So if no lookup has taken place for some reason the Grid from the previous lookup does not get cleared.
Thats..... Odd. Okay I'll track it down.

Thanks for the report.

mbridak, 2024-10-27:
Okay, I fixed the bug and pushed a new version. 24-10-27 to pypi.

Thank you again for the report. That bug must have been there from day one.

Cheers, Mike K6GTE

Bruce-N6BMW, 2024-10-27:
One other thing about Lookup. I've used both choices, QRZ and HamQTH. Does it record the name from the Lookup into the db or just the callsign?

mbridak, 2024-10-27:
Just the Grid, unless you have the general logging plugin loaded, then it'll save the name. 
But I don't consider the general logging plugin to be of any real use. It was just the first plugin made to see if the program would crash ;-)

closed 2024-10-27 by mbridak