mbridak/not1mm #212: flrig... I hate it. It hates me.

mbridak, 2024-11-10:
I've apparently neglected flrig for too long.

Long enough that it's causing problems. I'm not able to get my QMX to switch from digi to CW. And noted fldigi locked up while using
if with not1mm as soon as I switch over to using flrig. Others noted the applications disappear when using flrig. So I'm probably dumping a bunch of garbage to flrig and confusing the other kids in the playground.

mbridak, 2024-11-11:
@py4xx Okay pushed a change.
The flrig/fldigi/not1mm holy trinity works better here.
Update and give it a try please.

py4xx, 2024-11-12:
> @py4xx Okay pushed a change. The flrig/fldigi/not1mm holy trinity works better here. Update and give it a try please.

Hi Michael... 73
I updated to version 24.11.10, but I still have the same problem of closing not1mm, and it won't open again until I close FLDigi.
TNX

mbridak, 2024-11-12:
Open a terminal then type `not1mm` to launch it.
When it crashes or fails to open, it should print out why.

mbridak, 2024-11-12:
Add check for valid ipv4 address.

mbridak, 2024-11-15:
Are you still getting the crash? If so have you launched not1mm from a terminal so you can get the traceback output?

mbridak, 2024-11-17:
stale

closed 2024-11-17 by mbridak