mbridak/not1mm #223: Mode field of the export logs from CQWW CW looks funny...

awandahl, 2024-11-25:
**Describe the bug**
The export logs from the weekend's CQWW CW (ADIF and Cabrillo) looks funny for the mode field. Should be CW all the way. 

QSO:  3534 CW 2024-11-23 2349 SE0I          599 14     DM70DXMB      599 14    
QSO:  7026 CW 2024-11-23 2352 SE0I          599 14     IO4X          599 15    
QSO:  7031 CW 2024-11-23 2353 SE0I          599 14     OK5Z          599 15    
QSO:  7038 CW 2024-11-23 2355 SE0I          599 14     TM2Y          599 14    
QSO:  7047 CW 2024-11-23 2358 SE0I          599 14     OZ5W          599 14    
QSO:  7056 CW 2024-11-23 2359 SE0I          599 14     CN3A          599 33    
QSO:  3500 RPRT 2024-11-24 0002 SE0I          599 14     7S7V          599 14    
QSO:  3515 RPRT 2024-11-24 0005 SE0I          599 14     SP1NY         599 15    
QSO:  3519 RPRT 2024-11-24 0006 SE0I          599 14     S57KM         599 15    
QSO:  3520 RPRT 2024-11-24 0007 SE0I          599 14     UR4KWA        599 16    
QSO:  3527 RPRT 2024-11-24 0010 SE0I          599 14     OK7O          599 15    
QSO:  3540 RPRT 2024-11-24 0012 SE0I          599 14     OK1HEH        599 15    
QSO:  1816 RPRT 2024-11-24 0017 SE0I          599 14     RU1A          599 16    
QSO:  1824 RPRT 2024-11-24 0018 SE0I          599 14     LY2XW         599 15    
QSO:  1825 RPRT 2024-11-24 0023 SE0I          599 14     UZ6U          599 16    
QSO:  1822 CW 2024-11-24 0028 SE0I          599 14     OT7T          599 14    
QSO:  1836 CW 2024-11-24 0029 SE0I          599 14     OG1F          599 15    
QSO: 14023 14030000 2024-11-24 0034 SE0I          599 14     OH5Z          599 15    
QSO: 14026 14030000 2024-11-24 0035 SE0I          599 14     RZ1OA         599 16    
QSO:  7030 14030000 2024-11-24 0049 SE0I          599 14     YT6T          599 15    
QSO:  7001 14030000 2024-11-24 0052 SE0I          599 14     UW1M          599 16    
QSO:  7002 14030000 2024-11-24 0053 SE0I          599 14     HA3DX         599 15    
QSO:  7013 14030000 2024-11-24 0054 SE0I          599 14     YT7R          599 15    
QSO:  7014 14030000 2024-11-24 0055 SE0I          599 14     OL7D          599 15    
QSO:  7017 14030000 2024-11-24 0056 SE0I          599 14     DP6A          599 14    
QSO:  7021 14030000 2024-11-24 0057 SE0I          599 14     IB9R          599 15    
QSO:  7038 14030000 2024-11-24 0103 SE0I          599 14     9A5Y          599 15    
QSO:  7046 14030000 2024-11-24 0105 SE0I          599 14     IP2A          599 15    
QSO:  7052 14030000 2024-11-24 0107 SE0I          599 14     RT6A          599 16    
QSO:  3503 RPRT 2024-11-24 0110 SE0I          599 14     DM1T          599 14    
QSO:  3504 RPRT 2024-11-24 0110 SE0I          599 14     M6W           599 14    
QSO:  3506 RPRT 2024-11-24 0111 SE0I          599 14     R2FK          599 15    

**To Reproduce**
I can provide the database file if needed for troubleshooting.

**Desktop (please complete the following information):**
 - OS: xubuntu 24.04
 - Version latest

mbridak, 2024-11-25:
Oh, The rigctld responses must have gotten out of sync. The RPRT's are the start of the response line after it's received a command.

awandahl, 2024-11-25:
Thanks! I guess the database is contaminated so I will have to make a search/replace in the text file then? No problems. I have other things to report from my CQWW weekend in due time, mainly very positive experiences of not1mm. But I also had issues when changing bands and between RUN and S&P. That old problem where SSB switched itself in came back. Probably the root cause to the above? The last part of the contest I controlled band changing from the radio instead.

mbridak, 2024-11-25:
Pushed change rejecting RPRT.

mbridak, 2024-11-26:
Trying something new with rigctld. Should work better.

mbridak, 2024-12-04:
Closing as I believe it is resolved.

closed 2024-12-04 by mbridak