mbridak/not1mm #231: Add support for realtime scoreboard

mbridak, 2024-12-01:
Add support to post xml score info to contestonlinescore.com and hamscore.com.

Got the XML generating:

```xml
<?xml version="1.0"?>
<dynamicresults>
<contest>CQ-WW-CW</contest>
<call>K6GTE</call>
<class power="LOW" assisted = "ASSISTED" transmitter="ONE" ops="SINGLE-OP" bands="ALL" mode="CW" overlay="N/A"></class><club></club>
<soft>Not1MM</soft>
<version>24.11.27</version>
<qth>
  <cqzone>3</cqzone>
  <iaruzone>6</iaruzone>
  <arrlsection>ORG</arrlsection>
  <stprvoth>CA</stprvoth>
  <grid6>dm13at</grid6>
</qth>
<breakdown>
  <qso band="total" mode="ALL">5</qso>
  <mult band="total" mode="ALL" type="zone">5</mult>
  <mult band="total" mode="ALL" type="country">4</mult>
  <point band="total" mode="ALL">3</point>
</breakdown>
<score>27</score>
<timestamp>2024-12-01 06:22:25</timestamp>
</dynamicresults>
```

Just need to finish... Everything else.

awandahl, 2024-12-01:
contest.run is another scoreboard used extensively in Europe. Good idea to add scoreboard features. Missed that last weekend.

mbridak, 2024-12-02:
Yup, they even use the same xml tags.

mbridak, 2024-12-05:
@awandahl Okay, tested realtime scoring with contest.run. Seems to work. Now I just have to add it to all the contest plugins...

[image]

awandahl, 2024-12-06:
Great work! Thanks!

mbridak, 2024-12-21:
complete

closed 2024-12-21 by mbridak